greatscottgadgets/hackrf #1114: Environment Variable Possibly missing?

corcorj, 2022-07-01:
### Steps to reproduce
1.Open MSYS2 MinGW x64 from the Start Menu
2.Update the package repository
$ pacman -Syu
3.Install the required packages
$ pacman -Syu --needed base-devel mingw-w64-x86_64-toolchain mingw-w64-x86_64-cmake mingw-w64-x86_64-libusb mingw-w64-x86_64-fftw git
4.Clone the hackrf repository

$ git clone https://github.com/greatscottgadgets/hackrf.git

$ cd hackrf

$ mkdir host/build

$ cd host/build
5.Create the Makefiles
$ ${MINGW_PREFIX}/bin/cmake ../ -G "MSYS Makefiles" -DLIBUSB_INCLUDE_DIR=/lib -DFFTW_LIBRARIES=/lib

### Expected behaviour
Tell us what you expect should happen

I assume it would create the make file for installation of the hackrf tools 

### Actual behaviour
Tell us what happens instead

It output: bash: /bin/cmake: No such file or directory

### Version information
**Operating system**:
windows 11 pro 

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```
jcorc@Beast MSYS ~
$ pacman -Syu --needed base-devel mingw-w64-x86_64-toolchain mingw-w64-x86_64-cmake mingw-w64-x86_64-libusb mingw-w64-x86_64-fftw git
:: Synchronizing package databases...
 mingw32 is up to date
 mingw64 is up to date
 ucrt64 is up to date
 clang32 is up to date
 clang64 is up to date
 msys is up to date
:: There are 19 members in group mingw-w64-x86_64-toolchain:
:: Repository mingw64
   1) mingw-w64-x86_64-binutils  2) mingw-w64-x86_64-crt-git
   3) mingw-w64-x86_64-gcc  4) mingw-w64-x86_64-gcc-ada
   5) mingw-w64-x86_64-gcc-fortran  6) mingw-w64-x86_64-gcc-libgfortran
   7) mingw-w64-x86_64-gcc-libs  8) mingw-w64-x86_64-gcc-objc
   9) mingw-w64-x86_64-gdb  10) mingw-w64-x86_64-gdb-multiarch
   11) mingw-w64-x86_64-headers-git  12) mingw-w64-x86_64-libgccjit
   13) mingw-w64-x86_64-libmangle-git  14) mingw-w64-x86_64-libwinpthread-git
   15) mingw-w64-x86_64-make  16) mingw-w64-x86_64-pkgconf
   17) mingw-w64-x86_64-tools-git  18) mingw-w64-x86_64-winpthreads-git
   19) mingw-w64-x86_64-winstorecompat-git

Enter a selection (default=all):
:: Starting core system upgrade...
 there is nothing to do
:: Starting full system upgrade...
resolving dependencies...
looking for conflicting packages...

[... 414 lines cut ...]
$ git clone https://github.com/greatscottgadgets/hackrf.git
Cloning into 'hackrf'...
remote: Enumerating objects: 16226, done.
remote: Counting objects: 100% (1380/1380), done.
remote: Compressing objects: 100% (508/508), done.
remote: Total 16226 (delta 943), reused 1158 (delta 868), pack-reused 14846
Receiving objects: 100% (16226/16226), 39.62 MiB | 5.64 MiB/s, done.
Resolving deltas: 100% (11757/11757), done.

jcorc@Beast MSYS ~
$ cd hackrf

jcorc@Beast MSYS ~/hackrf
$ mkdir host/build

jcorc@Beast MSYS ~/hackrf
$ cd host/build

jcorc@Beast MSYS ~/hackrf/host/build
$ ${MINGW_PREFIX}/bin/cmake ../ -G "MSYS Makefiles" -DLIBUSB_INCLUDE_DIR=/lib -DFFTW_LIBRARIES=/lib
bash: /bin/cmake: No such file or directory

jcorc@Beast MSYS ~/hackrf/host/build
$ {MINGW_PREFIX}/bin/cmake ../ -G "MSYS Makefiles" -DLIBUSB_INCLUDE_DIR=/lib -DFFTW_LIBRARIES=/lib
bash: {MINGW_PREFIX}/bin/cmake: No such file or directory

corcorj, 2022-07-01:
I also have tried the command:

 cmake ../ -G "MSYS Makefiles" -DLIBUSB_INCLUDE_DIR=/usr/local/include/libusb-1.0/

stevewillson, 2022-07-01:
See comment in [#1107 ](https://github.com/greatscottgadgets/hackrf/issues/1107#issuecomment-1172444608), make sure there is a `$` in `${MINGW_PREFIX}`.
Check the contents of the `$MINGW_PREFIX` environment variable, should be `/mingw64`.

The cmake in `/mingw64/bin/cmake` supports the `-G "MSYS Makefiles"` generator option.

straithe, 2022-07-14:
@corcorj has @stevewillson's suggestion helped you out?

closed 2022-08-15 by straithe