greatscottgadgets/hackrf #1121: What is amplitude -inf dBfs ?

engantonio, 2022-07-21:
I am trying to generate a spoofer with hackRF but when i run it appear : 

 5.2 MiB / 1.006 sec =  5.2 MiB/second, amplitude -inf dBfs
 5.2 MiB / 1.006 sec =  5.2 MiB/second, amplitude -inf dBfs
 5.2 MiB / 1.007 sec =  5.2 MiB/second, amplitude -inf dBfs
 5.2 MiB / 1.009 sec =  5.2 MiB/second, amplitude -inf dBfs

Please what is "amplitude -inf dBfs" ?

martinling, 2022-07-21:
It's a bug, already filed as #902. The amplitude isn't calculated during TX. There is a PR to fix it at #1116 which I need to review.

closed 2022-07-21 by martinling

engantonio, 2022-07-21:
But 

> It's a bug, already filed as #902. The amplitude isn't calculated during TX. There is a PR to fix it at #1116 which I need to review.

But how can i solve this ?

martinling, 2022-07-21:
What do you need to solve? The message doesn't mean it's not transmitting. It just displays `-inf` because `hackrf_transfer` doesn't calculate the amplitude of the sample stream during TX. That amplitude display feature was intended to help with adjusting RX gains when receiving a signal. If you're transmitting a signal, the amplitude of the TX sample stream is whatever you generated it to be.

engantonio, 2022-07-21:
Both generating in linux or in windows the spoofer is not working.... but its transmiting...

ardyitjen, 2024-04-19:
is there any update? i got sampe problem with portapack h2

martinling, 2024-04-19:
The `-inf dBfs` message was fixed in #1116 and that fix was shipped in 2022.09.1 and subsequent releases of the host tools. In any case, this message is almost certainly nothing to do with your problem. It does not mean nothing is being transmitted, it was just a display bug.