greatscottgadgets/hackrf #1124: Issues with building firmware

THOMPSON832, 2022-07-24:
I have installed all dependencies and followed steps for building the firmware but still cannot get it built. After I cd into:

home/me/hackrf/firmware/hackrf_usb/build 

and execute:
 
sudo cmake /home/me/hackrf/firmware/hackrf_usb/build

this is the output:

-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /
-- The C compiler identification is unknown
CMake Error at CmakeLists.txt:26 (project):
    The CMake_C_Compiler:

        arm-none-eabi-gcc

    is not a full path and was not found in the PATH.

    Tell CMake where to find the compiler by setting either the environment variable "CC" or the CMake cache entry 
    CMAKE_C_COMPILER to the full path to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/me/hackrf/firmware/hackrf_usb/build/CMakeFiles/CMakeOutput.log"
See also "/home/me/hackrf/firmware/hackrf_usb/build/CMakeFiles/CMakeError.log"

I am running Dragon OS Focal R24 x86_64

martinling, 2022-07-24:
You're missing the required compiler. Since Dragon OS is Debian based I'm guessing you can `apt install gcc-arm-none-eabi`.

There's no need to run cmake under sudo, by the way.

THOMPSON832, 2022-07-24:
> You're missing the required compiler. Since Dragon OS is Debian based I'm guessing you can `apt install gcc-arm-none-eabi`.
> 
> There's no need to run cmake under sudo, by the way.

Thank you very much, I was able to build the firmware package and flash it to the hackrf. However it still made me use 'sudo' for 'cmake' and 'make' for the right permissions, does that have something to do with DragonOS?

miek, 2022-07-24:
> Thank you very much, I was able to build the firmware package and flash it to the hackrf. However it still made me use 'sudo' for 'cmake' and 'make' for the right permissions, does that have something to do with DragonOS?

When you ran cmake under sudo initially it would have created temporary build files as the root user, so then when you ran cmake again without sudo it didn't have permission to modify those files.

If you delete the build directory (with sudo), then re-create it and run cmake again (without sudo) it should all work.

THOMPSON832, 2022-07-24:
> When you ran cmake under sudo initially it would have created temporary build files as the root user, so then when you ran cmake again without sudo it didn't have permission to modify those files. [...]

Thank you! I'm still a bit new to linux, so your support is much appreciated!

straithe, 2022-07-25:
@THOMPSON832 have you successfully built the firmware now?

closed 2022-08-27 by straithe