greatscottgadgets/hackrf #1127: hackrf_sweep compiled separately after library install fails to get data from hackrf

mathewwe, 2022-07-26:
I am trying to compile a copy of hackrf_sweep in isolation as a first step before modifying the program for other purposes. 

### Steps to reproduce
1. Have functioning, installed hackrf library. hackrf_sweep functions correctly when run from the command line.
2. Compile a copy of hackrf_sweep.c in isolation, I tried both of the following compilation commands. The first is using flags I gathered from digging through cmake files, the second just links all the libraries necessary:
```
gcc -std=gnu90 -I /home/mathew/hackrf-2021.03.1/host/libhackrf/src -DTOOL_RELEASE=\"unknown\" -rdynamic hackrf_sweep.c -lhackrf -Wl,-rpath,/home/mathew/hackrf-2021.03.1/host/build/libhackrf/src: -lm -lfftw3f /home/mathew/hackrf-2021.03.1/host/build/libhackrf/src/libhackrf.so.0.6.0 -lusb-1.0 -lpthread -Wall

gcc hackrf_sweep.c -I /usr/local/include/libhackrf/ -lfftw3f -lhackrf -lm
```
Neither command yields errors.
3. Run the resulting program, `a.out`
### Expected behaviour
Ideally it should function exactly like hackrf_sweep.

### Actual behaviour
Data transfer fails with the output below. This is not the only program I have compiled using the installed libhackrf library, [this program functions correctly](https://gist.github.com/beurtschipper/a8c0b472392da212bf03b08f372a8f7d).

### Version information
**Operating system**: Ubuntu 20.04.4

**hackrf_info output:**
```
hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: <sn>
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: <pin1> <pin2>
```

### Output

```
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 6000 MHz
Stop with Ctrl-C
0 total sweeps completed, 0.00 sweeps/second

Couldn't transfer any data for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total sweeps: 0 in 1.00407 seconds (0.00 sweeps/second)
hackrf_close() done
hackrf_exit() done
exit
```

Are there some runtime dependencies that I'm missing? The radio briefly turns on (the RX light comes on,) but promptly shuts off. Any tips would be appreciated.

martinling, 2022-07-26:
I see from your paths that you're working from the 2021.03.1 release, but the `unknown` version in the `hackrf_info` output doesn't reflect that. I wonder if you have a different distribution-installed version of libhackrf installed in `/usr`, or one somewhere else in the path.

Output of `hackrf_info` on a freshly installed copy of the 2021.03.1 release package should display:
```
hackrf_info version: 2021.03.1
libhackrf version: 2021.03.1 (0.6)
Found HackRF
Index: 0
Serial number: ...
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: ...
```

I think you should add `-L/usr/local/lib` to your second command line, to ensure the program is linked against the version you are trying to use:

```gcc hackrf_sweep.c -I/usr/local/include/libhackrf/ -L/usr/local/lib -lfftw3f -lhackrf -lm```

With this command I had no problem building and running the `hackrf_sweep.c` from 2021.03.01 as an independent file.

Note that you can get the required compile parameters for libhackrf automatically with `pkg-config --cflags --libs libhackrf`.

The `-lfftw3f` and `-lm` you still need to add separately, as these are used independently by the `hackrf_sweep` code.

mathewwe, 2022-07-26:
After purging my system of anything remotely hackrf related, I reinstalled the hackrf release and the version is now showing correctly. That still wasn't enough to get the program to compile correctly. 

However, after that I ensured that my hackrf_sweep.c file was the latest version, and it began working properly. It's possible that I was using an old version of that file which was causing issues. Thanks for the help!

closed 2022-07-26 by mathewwe