greatscottgadgets/hackrf #1128: Hackrf one and SMA antena operating frequency

baegofa, 2022-07-27:
hello!

I bought a Hackrf one and an ANT 500 antenna for research purposes and did GPS spoofing. First of all, I plugged the ANT 500 antenna into the Hackrf one device and transmitted it in the 1575.42MHz frequency band using the Hackrf_transfer command, and the frequency transmission was possible successfully.

The curious thing here is that the operating frequency of the Hackrf one is 1 MHz to 6 GHz and the frequency of the ANT 500 antenna is 75 MHz to 1 GHz. The operating frequency of this ANT 500 antenna is 75 MHz to 1 GHz, so I wonder how it was able to transmit in a frequency band beyond that!

straithe, 2022-07-27:
Thank you for your question. The 75 MHz to 1 GHz description is an indicator of where the antenna performs best; it can work outside of that range but it often comes with reduced performance.

baegofa, 2022-08-01:
t

> 질문해주셔서 감사합니다. 75MHz ~ 1GHz 설명은 안테나가 가장 잘 작동하는 위치를 나타내는 지표입니다. 그 범위 밖에서도 작동할 수 있지만 종종 성능이 저하됩니다.

Thank you so much!

closed 2022-08-01 by baegofa