greatscottgadgets/hackrf #1130: HackRF doesn't see antenna

erd0spy, 2022-08-15:
### Steps to reproduce
1. HackRF one with multiple antenna(RTL-SDR Blog, ANT500, OpenSourceSDR Labs etc.)
2. GQRX 2.12, URH(latest)
3. Experiment: Try to receive FM, 433MHz or GSM signals at 9xxMHz

### Expected behaviour
I try to receive, capture, and send a signal with HackRF. I connect the SMA antenna to HackRF but the signal spectrum does not change, there is no difference between antenna or without antenna tests. I upgrade HackRF with the latest release but the info output seems to break.

### Actual behaviour
HackRF firmware upgrade:

```bash
dev@gsm:~/Desktop/hackrf/hackrf-2021.03.1/firmware-bin$ sudo hackrf_spiflash -w hackrf_one_usb.bin 
File size 35444 bytes.
Erasing SPI flash.
Writing 35444 bytes at 0x000000.
```

### Version information
**Operating system**: Ubuntu 20.04

**hackrf_info output:**

```bash
dev@gsm:~/Desktop/hackrf/hackrf-2021.03.1/firmware-bin$ sudo hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a06063c82547b45f
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x0057435c
```

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output

GQRX Output:

```bash
dev@gsm:~/Desktop/hackrf/hackrf-2021.03.1/firmware-bin$ gqrx
Controlport disabled
No user supplied config file. Using "default.conf"
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
gr::log :WARN: file_source0 - file size is not a multiple of item size
FM demod gain: 3.05577
Resampling audio 96000 -> 48000
IQ DCR alpha: 1.04166e-05
Using audio backend: auto
BookmarksFile is /home/dev/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

getDeviceList : Available input devices:
   "HackRF HackRF One 47b45f"
   "RFSPACE SDR-IQ Receiver"
   "RFSPACE SDR-IP Receiver"
   "RFSPACE NetSDR Receiver"
   "RFSPACE Cloud-IQ Receiver"
   "hw:Ensoniq AudioPCI,0"
   "default"
   "HackRF One #0 a06063c82547b45f"
   "RTL-SDR Spectrum Server"
   "Red Pitaya Transceiver Server"
   "Complex Sampled (IQ) File"
Loading configuration from: "default.conf"
Configuration file: "/home/dev/.config/gqrx/default.conf"
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using HackRF One with firmware 2021.03.1
IQ DCR samp_rate: 8e+06
IQ DCR alpha: 1.25e-07
Changing NB_RX quad rate: 96000 -> 8e+06
New antenna selected: "TX/RX"
********************
Gain name: "RF"
      min: 0
      max: 14
     step: 14
Gain name: "IF"
      min: 0
      max: 40
     step: 8
Gain name: "BB"
      min: 0
      max: 62
     step: 2
********************
No change in output device:
  old: 
  new: 
IQ DCR samp_rate: 8e+06
IQ DCR alpha: 1.25e-07
Requested sample rate: 8000000
Actual sample rate   : "8000000.000000"
New FFT rate: 25 Hz
Requested bandwidth: 0 Hz
Actual bandwidth   : 6e+06 Hz
setFreqCorr : 0 ppm
New LNB LO: 0 Hz
New mode index: 2
Filter preset for mode 2 LO: -5000 HI: 5000
Generating taps for new filter   LO:-5000   HI:5000 TW:2000   Taps: 115
setFftRate to "25 fps"
New FFT rate: 25 Hz
New FFT rate: 25 Hz
setFftSize to "8192"
New FFT rate: 25 Hz
Generating taps for new filter   LO:-5000   HI:5000 TW:2000   Taps: 115
New FFT rate: 25 Hz
New FFT rate: 25 Hz
No audio FFT data.
No audio FFT data.
No audio FFT data.
No audio FFT data.
No audio FFT data
```

[image]

Example RTL SDR Output with same antenna and same conditions:

```bash
dev@gsm:~/Desktop/hackrf/hackrf-2021.03.1/firmware-bin$ gqrx
Controlport disabled
No user supplied config file. Using "default.conf"
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
gr::log :WARN: file_source0 - file size is not a multiple of item size
FM demod gain: 3.05577
Resampling audio 96000 -> 48000
IQ DCR alpha: 1.04166e-05
Using audio backend: auto
BookmarksFile is /home/dev/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

Found Rafael Micro R820T tuner
getDeviceList : Available input devices:
   "Realtek RTL2838UHIDIR SN: 00000001"
   "RFSPACE SDR-IQ Receiver"
   "RFSPACE SDR-IP Receiver"
   "RFSPACE NetSDR Receiver"
   "RFSPACE Cloud-IQ Receiver"
   "hw:Ensoniq AudioPCI,0"
   "default"
   "Generic RTL2832U OEM :: 00000001"
   "RTL-SDR Spectrum Server"
   "Red Pitaya Transceiver Server"
   "Complex Sampled (IQ) File"
Loading configuration from: "default.conf"
Configuration file: "/home/dev/.config/gqrx/default.conf"
Crash guard triggered! 

Launching I/O device editor
firstTimeConfig
CIoConfig : Available output devices:
    0 : "ES1371/ES1373 / Creative Labs CT2518 (Audio PCI 64V/128/5200 / Creative CT4810/CT5803/CT5806 [Sound Blaster PCI]) Analog Stereo"
saveConfig
Loading configuration from: "/home/dev/.config/gqrx/default.conf"
Configuration file: "/home/dev/.config/gqrx/default.conf"
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[R82XX] PLL not locked!
IQ DCR samp_rate: 1.024e+06
IQ DCR alpha: 9.76562e-07
Changing NB_RX quad rate: 96000 -> 1.024e+06
New antenna selected: "RX"
********************
Gain name: "LNA"
      min: 0
      max: 49.6
     step: 0.5
********************
No change in output device:
  old: 
  new: 
[R82XX] PLL not locked!
IQ DCR samp_rate: 1.8e+06
IQ DCR alpha: 5.55555e-07
Changing NB_RX quad rate: 1.024e+06 -> 1.8e+06
Requested sample rate: 1800000
Actual sample rate   : "1800000.000000"
New FFT rate: 25 Hz
Requested bandwidth: 0 Hz
Actual bandwidth   : 0 Hz
setFreqCorr : 0 ppm
New LNB LO: 0 Hz
New mode index: 2
Filter preset for mode 2 LO: -5000 HI: 5000
Generating taps for new filter   LO:-5000   HI:5000 TW:2000   Taps: 115
setFftRate to "25 fps"
New FFT rate: 25 Hz
New FFT rate: 25 Hz
setFftSize to "8192"
New FFT rate: 25 Hz
Generating taps for new filter   LO:-5000   HI:5000 TW:2000   Taps: 115
New mode index: 4
FM demod gain: 0.509296
Filter preset for mode 4 LO: -80000 HI: 80000
Generating taps for new filter   LO:-80000   HI:80000 TW:32000   Taps: 19
Allocating 15 zero-copy buffers
New FFT rate: 25 Hz
New FFT rate: 25 Hz
No audio FFT data.
No audio FFT data.
No audio FFT data.
No audio FFT data.
No audio FFT data.
No audio FFT data.
No audio FFT data.
No audio FFT data.
saveSettings *** FIXME_ SQL on/off
New FFT rate: 25 Hz
dev@gsm:~/Desktop/hackrf/hackrf-2021.03.1/firmware-bin$ 
```

[image]

erd0spy, 2022-08-15:
Extra:

[image]

straithe, 2022-08-15:
Interesting antenna! Can you please remove it from your HackRF and show me the inside of the connector? Or provide me a link to where you purchased this antenna?

straithe, 2022-09-14:
I'm closing this as there hasn't been a response in 30 days. Please re-open this issue or open a new one if you still need assistance.

closed 2022-09-14 by straithe