greatscottgadgets/hackrf #1134: Hack RF USB Not connecting to computer.

npepper1, 2022-08-17:
I have a HackRF+Portapack 1.  I am trying to connect it to my computer via USB, but when I plug it into the USB, the light does not come on and the device does not show as connected on my computer.  The portapack did not come with a battery only USB connection. 

USB led does not light up and the device is not discoverable by Windows, MacOS or Linux. 
It is not shown in the Windows 10 Device Manager, or in the MacOS System Report, or found in Linux.
[image]
[image]
[image]
[image]

straithe, 2022-08-17:
What happens when you try to plug in the HackRF without the Portapack attached?

npepper1, 2022-08-17:
The same result as above.

straithe, 2022-08-17:
How many different USB cables have you tried?

closed 2022-08-17 by npepper1