greatscottgadgets/hackrf #1136: Hackrf one Spectrum Analyzer for Windows 10

Shaneel123, 2022-08-22:
### Steps to reproduce -
1. Windows 7+ x64 required - I HAVE WINDOWS 10
2. Install Java JRE 64bit v1.8+ - DONE
3. [Download the latest version of Spectrum Analyzer](https://github.com/pavsa/hackrf-spectrum-analyzer/releases) and unzip - DONE
4. Install HackRF as a libusb device - NOT SURE EXACTLY WHAT THIS MEANS
5. [Download Zadig](https://github.com/pavsa/hackrf-spectrum-analyzer/blob/master/src/hackrf-sweep/lib/zadig_2.2.exe) and run - CAME WITH THE PACKAGE I DOWN LOADED
6. Goto Options and check List All Devices - DONE
7. Find "HackRF One" and select Driver "WinUSB" and click install - DONE
8. Install (if you don't have one installed) [Java JRE for Windows x64]  - DONE
9. (http://www.oracle.com/technetwork/java/javase/downloads/jre8-downloads-2133155.html) - DONE
10. Run "hackrf_sweep_spectrum_analyzer_windows.cmd" - DONE
11. Shows Hackrf one as disconnected. i have tried many different configurations and start up options but its always disconnected

### Expected behaviour
I thought the devices would boot up and start reading the default frequency range, but i dont see nothing

### Actual behaviour
Graph and options for the spectrum analyser appear but no continuous stream of data is detected

### Version information
Windows 10 PC
I dont know how to check what my Hackrf one is?

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

martinling, 2022-08-22:
The hackrf-spectrum-analyzer software isn't part of the HackRF project, it's a third party program maintained at https://github.com/pavsa/hackrf-spectrum-analyzer.

It hasn't been updated in some years and there are already several issues open saying that it doesn't work on Windows 10 or with recent HackRF versions.

You may want to try QSpectrumAnalyzer instead: https://github.com/xmikos/qspectrumanalyzer

closed 2022-08-22 by martinling

Shaneel123, 2022-08-22:
Thank you for the Guidance!!!!