greatscottgadgets/hackrf #1140: Cannot write custom firmware to CPLD

braedinski, 2022-08-27:
I've been trying to run a custom `default.xsvf` to expose the ADC pins to the nearby P30 `BANK2_AUX` header. Unfortunately, my modifications don't seem to have any effect. Since I'm "mirroring" pins, I've made sure to prevent my changes from being optimised away by using the `equivalent_register_removal` attribute in the `top.vhdl` file.

My constraint file is as follows:
```
NET "CODEC_CLK"  LOC = "P23" ; 
NET "CODEC_X2_CLK"  LOC = "P27" ; 
NET "DADUP<0>"  LOC = "P13" ; 
NET "DADUP<1>"  LOC = "P12" ; 
NET "DADUP<2>"  LOC = "P11" ; 
NET "DADUP<3>"  LOC = "P10" ; 
NET "DADUP<4>"  LOC = "P9" ; 
NET "DADUP<5>"  LOC = "P8" ; 
NET "DADUP<6>"  LOC = "P7" ; 
NET "DADUP<7>"  LOC = "P6" ; 
NET "DA<0>"  LOC = "P43" ; 
NET "DA<1>"  LOC = "P42" ; 
NET "DA<2>"  LOC = "P41" ; 
```
I believe that these pins are assigned to the `B2AUX1` through to `B2AUX8` pads.

Is it still possible to flash new firmware using the included instructions? I've built everything successfully using Xilinx ISE 14.7 on Ubuntu 16.04 but can't see the ADC pins being mirrored on my oscilloscope. After flashing the lights blink and there is no error thrown.

martinling, 2022-08-27:
Since PR #600, the firmware image carries its own copy of the CPLD bitstream, and will load it into the CPLD's configuration SRAM at startup, so the bitstream in the CPLD flash is effectively ignored.

So if you're using `hackrf_cpldjtag` to write your new `xsvf` file, that will now be ignored. If you want to run a custom CPLD bitstream you'll need to patch the firmware in either of the following ways, whichever you prefer:

- Remove the call to `cpld_jtag_sram_load` from `main` in `hackrf_usb.c`. This will stop the firmware from replacing the bitstream stored in the CPLD's flash, so your bitstream written with `hackrf_cpldjtag` will be used.

- Replace `firmware/cpld/sgpio_if/default.xsvf` with your new version, so that the firmware loads your bitstream instead.

Then rebuild the firmware and flash the patched version with `hackrf_spiflash`.

braedinski, 2022-08-27:
Thank you! I remember reading that the CPLD bitstream was now included in the firmware image but I didn't understand why the `hackrf_cpldjtag` utility would still be included. Perhaps a message should be included in the utility itself?

martinling, 2022-08-28:
I think initially `hackrf_cpldjtag` was retained because because people might still need it to work with older versions or with customised firmware. Adding a message would make sense though. I've added issue #1141 to track this.

closed 2022-08-28 by martinling