greatscottgadgets/hackrf #1152: Different HackRF One-s give inconsited behaviour

VasilyRakche, 2022-09-05:
### What would you like to know?

Hi, I have two HackRF One-s, but the behaviour is different between the two. Running the
[t1_checking_signals.zip](https://github.com/greatscottgadgets/hackrf/files/9487942/t1_checking_signals.zip), can be seen on the picture:
[image]  
With one HackRF I get the following:  
[image]  
And with the second one I get the following:  
[image]

What is strange is the DC offset is different, I also noticed that the noise floor is a bit higher. What do you think?

straithe, 2022-09-05:
Where did you get each HackRF from? Can you please post pictures of both?

straithe, 2022-12-06:
To help me diagnose this can you please repeat the test with the RF gain set to 0. Ideally the test would be done with a cabled RF signal generator.

closed 2023-04-02 by straithe