greatscottgadgets/hackrf #1158: [Question]:

jadams454, 2022-09-14:
### What would you like to know?

Hi,

HackRF newbie here. I am trying to run some basic code here, receive FM station. When I used a Soapy HackRF Sink and a QT GUI Frequency Sink, I only see one tone at the frequency that i specify for the HackRF. Nothing else around it. The lights on the side of the HackRF are on and RX is lit.

I am running Ubuntu, and when I run the hackrf_info command legit looking data shows up. All packages are updated.

I have updated the firmware to: 2021.03.1

What am I doing wrong?

[image]

[image]

[image]

straithe, 2022-09-14:
Please review the GNU Radio tutorials: https://wiki.gnuradio.org/index.php/Tutorials

jadams454, 2022-09-14:
> Please review the GNU Radio tutorials: https://wiki.gnuradio.org/index.php/Tutorials

Will do, in the meantime is there an obvious gotcha here? Thanks!

straithe, 2022-09-14:
You are hearing a tone so that seems fine. What are you hoping to have happen?

jadams454, 2022-09-14:
See fm radio stations around the spectrum, a tone comes up where I set the
center frequency but no where else.

jadams454, 2022-09-14:
> The spike/tone is the [DC offset](https://hackrf.readthedocs.io/en/latest/faq.html#bigspike).
> 
> 
> 
> You may need to turn up some gains in the source block. I assume you have a suitable antenna connected?

The antenna that came with it. I also saw that in the FAQ. I updated firmware. Not sure if this could be a version mismatch and how to tell?

[image]

jadams454, 2022-09-14:
> The spike/tone is the [DC offset](https://hackrf.readthedocs.io/en/latest/faq.html#bigspike).
> 
> 
> 
> You may need to turn up some gains in the source block. I assume you have a suitable antenna connected?

Also changing gains didn't help.

dmaltsiniotis, 2022-09-14:
@jadams454 
For what it's worth, I've replicated the simple flow graph (on Windows) and it works as expected:
[image]
It could be that there's something off with the Soapy -> HackRF layer on your Ubuntu install. What I would recommend is installing the gr-osmocom (`apt-get install gr-osmosdr`) blocks and see if the problem persists. That will at least narrow it down:
[image]

straithe, 2022-11-22:
@jadams454 are you still looking for assistance or may I close this issue?

jadams454, 2022-11-22:
All set thanks

closed 2022-11-23 by straithe