greatscottgadgets/hackrf #1159: [Bug Report]: hardware sync output inconsistent in TX mode

mossmann, 2022-09-16:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

When using a HackRF One to produce a synchronization pulse for another HackRF One that is waiting for hardware sync input, the output pulse timing is inconsistent with respect to TX samples being transmitted by the HackRF One that is producing the sync pulse.

If using one HackRF One in TX mode and another in RX mode, this can be worked around by having the receiving HackRF One, not the transmitting HackRF One, produce the sync pulse or by using an external source to deliver a sync pulse to both HackRF Ones.

If using two HackRF Ones in TX mode, this can be worked around by using an external source to deliver a sync pulse to both HackRF Ones.

### What are the steps to reproduce this?

- Connect CLKOUT of receiving HackRF One to CLKIN on transmitting HackRF One with SMA cable.
- Connect wire from P28 pin 15 of transmitting HackRF One to P28 pin 16 of receiving HackRF One.
- Connect RF port of transmitting HackRF One through a 20 dB attenuator to the RF port of receiving HackRF One.
- Execute commands:
```
hackrf_clock -d <rx serial number> -o 1
hackrf_transfer -d <rx serial number> -a 0 -l 20 -g 20 -r rx.cs8 -n 10000000 -H 1
```
- In another terminal execute:
```
hackrf_transfer -d <tx serial number> -a 0 -x 40 -c 127 -n 600000
```
Repeat test and observe variation in timing of received signal in rx.cs8 file.

mossmann, 2022-09-16:
I observed this issue with:
- host code and firmware from 2b1aa54
- host code from release 2021.03.1 and firmware from 2b1aa54

I could not reproduce this issue with host code and firmware both from release 2021.03.1, so it seems that the root cause is in a firmware change since that release.

martinling, 2022-09-20:
I think this was broken by #982, and that a bisect would probably find the breaking commit is c0d0cd2a, which is where the M0 firsts start sending zeros to the CPLD if there isn't data available in the buffer yet. This doesn't get counted as an underrun in the M0 stats when it happens at the start, because we consider waiting for the first data to be normal. But it does mean that the sync output will no longer align with sample timings.

Before that change, TX would proceed with whatever garbage was in the buffer, but once the buffer started to be filled with the correct samples they would be sent on-time, relative to the trigger output.

Having both HackRFs triggered from an external source fixes this, because it gives the host time to fill the buffer on the transmitting HackRF before TX is triggered.

martinling, 2022-09-20:
Possible solution: at TX startup, have the M4 not run `baseband_streaming_enable` until the first two 16KB transfers have arrived from the host, meaning that the M0 has a full buffer ready to transmit.

closed 2022-09-21 by mossmann