greatscottgadgets/hackrf #1160: [Tech Support]:

waldo545, 2022-09-16:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

After a great exposition of Rf, hackrf stopped receiving. Changed U13 and hoped to fix the device

### What outcome actually happened?

What happened was that it didn't received anything. 
I have observed that in test point 52 there is 3,5V and in test point 49 0V, so no voltage in U13 and no antenna switch in U14

### What operating systems are you seeing the problem on?

Debian Bullseye

### What is the output of ```hackrf_info```?

eduardo@waldo2:~$ hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000017c467dc3962a4c3
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x00544756

### Are you using any third-party software?

GNU Radio, Gqrx

### Are you using any third-party hardware?

No

mmm545, 2022-09-16:
Does test point P49 become high when turning on the RF AMP option in your software?

waldo545, 2022-09-17:
Thanks for the comment,
That was the point, I had not enabled the rf amp, I think I'm too old for this rock and roll.
Nevertheless sensitivity is very low, so I'll have to keep looking the other circuitry phases for if another MGA or SKY has passed away.
I'll be in touch with the progress, if any.
Thanks again mmm545
waldo

waldo545, 2022-09-17:
Sorry, issue is closed

closed 2022-09-18 by waldo545