greatscottgadgets/hackrf #1170: [Tech Support]: HackRF Not Being Seen

KyleKaminky, 2022-09-21:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

I just installed Ubuntu brand new on my laptop and then installed gnuradio-companion, hackrf and gqrx. I was hoping to run hackrf_info and see my hackrf connected. I then tried installing hackrf from source.

### What outcome actually happened?

I am not seeing any connection and the usb light on my hackrf is not lighting up. When I plug it into my mac I do see the USB led illuminate and I am able to get results from hackrf_info

### What operating systems are you seeing the problem on?

Ubuntu 22.04.1

### What is the output of ```hackrf_info```?

hackrf_info version: git-e623c7ae
libhackrf version: git-e623c7ae (0.7)
No HackRF boards found.

### Are you using any third-party software?

I installed gnuradio with "sudo apt-get install gnuradio"
I installed gr-osmosdr with "sudo apt install gr-osmosdr"
I installed gqrx with "sudo apt install gqrx-sdr"

### Are you using any third-party hardware?

No

straithe, 2022-09-21:
Thank you for reaching out. Have you tried a different USB cable?

KyleKaminky, 2022-09-21:
Yes I have

straithe, 2022-09-21:
I suggest trying ~5 different cables if you can. You are doing the right thing by checking `hackrf_info` after each cable. Another thing you could check is lsusb at the same time. There have been instances where the 5th cable works.

KyleKaminky, 2022-09-21:
Holy crap haha. I honestly thought that was a little ridiculous, and what do you know.... the fifth cable worked!!! It also seems like the last cable I tried has the highest quality. Thank you for your help!

straithe, 2022-09-22:
Happy to help! ☺️

closed 2022-09-22 by straithe