greatscottgadgets/hackrf #1173: External clock does not work[Tech Support]:

hamidghafi, 2022-09-22:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

When I use hackrf_debug --si5351c -n 0 -r, I expect 0x01 as TCXO is connected. 

### What outcome actually happened?

When I use hackrf_debug --si5351c -n 0 -r, I see 0x50 as TCXO. As the issue is like #1099 I noticed that the clock generator chip is si5351AMF which is likely the source of the problem. As I have to use multi SDRs with the synched clock. Is there any solution for that? for example, does si5351AMF receive higher frequencies of clock-in signals? or is it possible to simply buy a si5351c or b and replace it? 

### What operating systems are you seeing the problem on?

Linux, windows

### What is the output of ```hackrf_info```?

hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc3345a5c3
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x004b435e

### Are you using any third-party software?

No.

### Are you using any third-party hardware?

No.

mossmann, 2022-09-22:
If your board is populated with Si5351A, the best way to add CLKIN support is to replace that part with Si5351C.

The only other way to achieve external clock input with Si5351A would be to remove crystal X1 and bodge a connection from CLKIN, through a DC blocking capacitor, and into pin 1 of the Si5351A. You may also need a voltage divider as the maximum input voltage on pin 1 is 1 V peak-to-peak. See "Applying a Reference Clock at XTAL Input" in the Si5351A data sheet. If you were to do this you would lose the ability to operate your HackRF One without an external clock reference. Additionally your clock reference would need to have a frequency of 25 MHz unless you modify the firmware to support a different frequency (which must be between 10 MHz and 40 MHz) on the XTAL (pin 1) input.

mossmann, 2022-09-22:
I should also mention that every HackRF One made Great Scott Gadgets supports external clock input. We work hard to develop and support the HackRF project, and we truly appreciate it when folks support us by purchasing genuine Great Scott Gadgets products from any of our resellers: https://greatscottgadgets.com/wheretobuy/

mossmann, 2022-09-23:
Complicating things further, it isn't 100% clear that your HackRF is populated with Si5351A, although I think that is very likely.

The part number specified in the HackRF One design is Si5351C-B-GM. I suspect that the part installed on your board is Si5351A-A-GM. The first "A" in Si5351A-A-GM is the "variant" that indicates the number and type of clock inputs. The second "A" is the "product revision". All part numbers listed in the current data sheet are product revision B as revision A is now obsolete.

The contents of register address `0` that you observed with `hackrf_debug --si5351c -n 0 -r` show that the part on your board is revision A but do not indicate the variant. Similarly the "AMF" marking on your chip's package indicates that it is revision A, not that it is variant A.

I think your part is probably Si5351A-A-GM, but it might be Si5351C-A-GM or even Si5351B-A-GM. Only the C variant (Si5351C) supports an external clock input separate from the crystal input. The only method I know of to determine that it is not the C variant is to observe that a 10 MHz signal on CLKIN cannot be detected.

closed 2022-11-22 by straithe