greatscottgadgets/hackrf #1177: Final TX buffer is repeated while hackrf_transfer exits

mikotaFull, 2022-09-24:
Are you still experiencing this issue?

_Originally posted by @straithe in https://github.com/greatscottgadgets/hackrf/issues/660#issuecomment-771995408_

mikotaFull, 2022-09-24:
hello,
I tried what miek recommends and I still get the signal repeated 160 times. I use gr-air-modes to decode the signal and as you can see on the picture it is repeated 160 times.

Here is the sequence of commands I use:
ADSB_Encoder.py 0xABCDEF 12.34 56.78 9999.0
dd if=Samples.iq8s of=Samples_256K.iq8s bs=4k seek=63
cp Samples_256K.iq8s Samples_256K_padded.iq8s
dd if=/dev/zero bs=16384 count=16 >>samples_256K_padded.iq8s
hackrf_transfer -t Samples_256K_padded.iq8s -f 1090000000 -s 2000000 -x 10 -a 1

it is right?
moreover I notice that the file Samples_256K_padded.iq8s is repeated twice which does not seem normal to me.
please could you help me to solve this problem?
Thank you

[image]

mikotaFull, 2022-09-24:
hello,
I tried what miek recommends in the  [issue]( https://github.com/greatscottgadgets/hackrf/issues/660
) 
[image]
and I still get the signal repeated 160 times. I use gr-air-modes to decode the signal and as you can see on the picture it is repeated 160 times.

Here is the sequence of commands I use:
ADSB_Encoder.py 0xABCDEF 12.34 56.78 9999.0
dd if=Samples.iq8s of=Samples_256K.iq8s bs=4k seek=63
cp Samples_256K.iq8s Samples_256K_padded.iq8s
dd if=/dev/zero bs=16384 count=16 >>samples_256K_padded.iq8s
hackrf_transfer -t Samples_256K_padded.iq8s -f 1090000000 -s 2000000 -x 10 -a 1

it is right?
moreover I notice that the file Samples_256K_padded.iq8s is repeated twice which does not seem normal to me.
please could you help me to solve this problem?
Thank you

martinling, 2022-09-24:
Hi @mikotaFull, a lot of things have been fixed in the HackRF master branch recently, are you testing with current git master for the firmware and host software? If not, please could you try that?

mikotaFull, 2022-09-26:
Hi @martinling, 

No, I didn't.
I never updated the firmware directly from the source file of the master branch. After reading the instructions I have 2 questions:

1) for the gcc toolchain do I have to install both executables for linux?
    xjf gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2
    xjf gcc-arm-none-eabi-10.3-2021.10-aarch64-linux.tar.bz2
In the readme they put both, I don't know if it's an example or I really need both

2) to install libopencm3 the command sequence is the following ?:
   $ git clone https://github.com/libopencm3/libopencm3.git aClone
   $ cd aClone
   $ git submodule init
   $ git submodule update

Thanks

martinling, 2022-09-26:
Hi @mikotaFull,

1. You only need the one appropriate to the machine you're building on - `x86_64` if you're on a PC of some sort, `aarch64` for Apple M1 or other ARM machines.

2. libopencm3 is already a submodule in the hackrf repository, so you just need to run `git submodule init` and `git submodule update` within the hackrf repository.

mikotaFull, 2022-09-26:
Hi @martinling,

understood, thank you.

I have another question, I am following the [instructions on github ](https://github.com/greatscottgadgets/hackrf/tree/master/firmware) and they say to execute $hackrf_spiflash -w hackrf_usb.bin
but I also looked at the [tutorial on youtube](https://www.youtube.com/watch?v=JMfDd_fZiOU&t=1695s) and Michael says to run $hackrf_spiflash -w hackrf_one_usb_rom_to_ram.bin
which one should I run? 
Are they different because one is the realease and the other is not?

straithe, 2022-09-26:
The tutorials are from 2014. Please follow the instructions on GitHub as those have been updated and tested more recently.

martinling, 2022-09-26:
You should flash `hackrf_usb.bin`. The instructions in that video are out of date for more recent releases.

mikotaFull, 2022-09-27:
Hello,
I unloaded the gcc-arm-none-eabi and then followed the readme instructions to extract it, install it and add it to the PATH. Everything looked fine as you can see in the picture below.

[image]

Then I went to the hackrf/firmware/hackrf_usb file, ran the git init and git update commands, executed mkdir build, cd build, cmake .. and this last instruction gives me errors, it says that the gcc is not in the PATH...
[image]
what's wrong?

martinling, 2022-09-27:
Are those two screenshots from the same terminal session? That `PATH` setting will only apply to the current shell.

mikotaFull, 2022-09-27:
Hi, 
now cmake works but when I do the make a have this output:
Inspiron-3793:~/Workarea/hackrf-master/firmware/hackrf_usb/build$ make
Scanning dependencies of target libopencm3_hackrf_usb
[  1%] Creating directories for 'libopencm3_hackrf_usb'
[  1%] No download step for 'libopencm3_hackrf_usb'
[  1%] No patch step for 'libopencm3_hackrf_usb'
[  2%] No update step for 'libopencm3_hackrf_usb'
[  3%] No configure step for 'libopencm3_hackrf_usb'
[  4%] Performing build step for 'libopencm3_hackrf_usb'
make[3]: *** No targets specified and no makefile found.  Stop.
make[2]: *** [CMakeFiles/libopencm3_hackrf_usb.dir/build.make:112: libopencm3_hackrf_usb-prefix/src/libopencm3_hackrf_usb-stamp/libopencm3_hackrf_usb-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:99: CMakeFiles/libopencm3_hackrf_usb.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
Inspiron-3793:~/Workarea/hackrf-master/firmware/hackrf_usb/build$ 

Before the build I made:
Inspiron-3793:~/Workarea$ cd hackrf-master
Inspiron-3793:~/Workarea/hackrf-master$ git init
Initialized empty Git repository in /home/mmmmm/Workarea/hackrf-master/.git/
Inspiron-3793:~/Workarea/hackrf-master$ git submodule init
Inspiron-3793:~/Workarea/hackrf-master$ git submodule update

mikotaFull, 2022-09-27:
the libopencm3 folder in the hackrf_master folder it was so I did : 
$ git clone https://github.com/libopencm3/libopencm3.git libopencm3
$ cd libopencm3
$ git submodule init
$ git submodule update

 
then when I did the make

Dependency analysis of the libopencm3_hackrf_usb target
[1%] Creation of directories for 'libopencm3_hackrf_usb'.
[1%] No download steps for 'libopencm3_hackrf_usb'.
[1%] No patch steps for 'libopencm3_hackrf_usb'.
[2%] No update steps for 'libopencm3_hackrf_usb'.
[3%] No configuration steps for 'libopencm3_hackrf_usb'.
[4%] Execute build step for 'libopencm3_hackrf_usb'.
  GENHDR stm32/f0
  GENHDR stm32/f1
  GENHDR stm32/f2
  GENHDR stm32/f3
  GENHDR stm32/f4
  GENHDR stm32/f7
  GENHDR stm32/l0
  GENHDR stm32/l1
  GENHDR stm32/l4
  GENHDR stm32/g0
  GENHDR stm32/g4
  GE
...
 AR libopencm3_pac55xx.a
[4%] No installation steps for 'libopencm3_hackrf_usb'.
[5%] Finished 'libopencm3_hackrf_usb'.
[5%] Build target libopencm3_hackrf_usb
Analysis of the dependencies of the target hackrf_usb_m0.elf
[5%] Construction of the ASM object CMakeFiles/hackrf_usb_m0.elf.dir/sgpio_m0.s.obj
/home/mikaela/Workarea/hackrf-master/firmware/hackrf_usb/sgpio_m0.s: Assembler messages:
/home/mikaela/Workarea/hackrf-master/firmware/hackrf_usb/sgpio_m0.s:569: Warning: ignores redefinition of the 'prev' registry alias.
[6%] Binding of the ASM executable hackrf_usb_m0.elf
/usr/bin/gcc-arm-none-eabi-10.3-2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld : cannot open linker script file libopencm3_lpc43xx_m0.ld : No such file or directory
collect2 : error : ld returned 1 exit status
make[2]: *** [CMakeFiles/hackrf_usb_m0.elf.dir/build.make:76: hackrf_usb_m0.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:126: CMakeFiles/hackrf_usb_m0.elf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

martinling, 2022-09-27:
You didn't need to run `git init` in the hackrf repository, this would have re-initialized the repository and stopped the submodules working. You may need to make a fresh clone at this point.

You just need to run `git submodule init` and `git submodule update` from the hackrf repository, doing that will take care of cloning libopencm3 and checking out the correct version.

mikotaFull, 2022-09-27:
Hi @martinling,
I made a fresh clone and followed the instructions step by step and everything worked. 
Once that was done I also tried the workaround that @miek suggested and it works.
Thanks so much for your support and patience.

straithe, 2022-09-27:
You're welcome! We are all happy to help. Now that it seems your issue is resolved, may I close this issue?

martinling, 2022-09-28:
You should also find that with the new version, you no longer need the workaround to solve your original issue.

The problem with the firmware recycling previous samples from the TX buffer was fixed in PR #982.

The need to fill a 256KB block of samples was removed in PR #1156.

So provided you've updated your host software to git master as well as the firmware, then referring to your original example from #660, you should just be able to run:

`hackrf_transfer -t Samples.iq8s -f 1090000000 -s 2000000 -x 0`

without needing to pad the samples out to 256KB.

mikotaFull, 2022-09-28:
Hello 

@martinling thanks for the info, I was not aware of all these improvements and I thought that the only way to make the code work is with the workaround.
@straithe yes, you can close the issue.

Once again thanks to both of you :)

closed 2022-09-28 by straithe