greatscottgadgets/hackrf #1192: [Bug Report]: Hackrf_info without sudo on latest release

alphafox02, 2022-09-29:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

Very well could be just me, but I notice after replacing libhackrf/firmware using the lastest 2022 release, there's about a 15 second delay between plugging in the hackrf and being able to run hackrf_info without sudo.

### What are the steps to reproduce this?

Updated to the last release, lib and firmware on 20.04. Unplug the hackrf and plug it back in. Continue to rerun hackrf_info and see if you get the following,

`hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
Found HackRF
Index: 0
hackrf_open() failed: Access denied (insufficient permissions) (-1000)
`

After about 15 seconds I then get this, which is what I expected, and then all seems okay. I just don't recall waiting as long before, this is also after double checking udev rules and reloading. 

`hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
Found HackRF
Index: 0
Serial number: xxxxxxxxxxxxxxxxxxxxxxxxxxx
Board ID Number: 2 (HackRF One)
Firmware Version: 2022.09.1 (API:1.06)
Part ID Number: xxxxxxxxxxxxxxxx
Hardware Revision: r6
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
`

martinling, 2022-09-29:
That's very odd. Are you sure you don't have some other software trying to use the device at the same time?

It's hard to see how any changes in the release could be causing this, since device permissions are set by udev, not by any HackRF code. The udev rules supplied haven't changed since 2015.

If this is happening due to something in the release, it would be helpful to work out whether it's on the firmware or the host side.

Could you put the firmware back to the 2021 version, whilst keeping the 2022 host software, and check if the delay goes away? Or the other way around, with 2022 firmware and 2021 host software?

alphafox02, 2022-09-29:
Left the libs/host software to the last and reverted to 2021.03.1 firmware on the hackrf. Behavior is the same, about a 15 second delay before hackrf_info will spit out full results without using sudo.

Next I upgraded the firmware back to 2022 and downgraded the libs/software to 2021 (making sure .so's were gone). Behavior was the same except this time hackrf_info spit this out after about 15 seconds of repeated running it without sudo.

hackrf_info version: 2021.03.1
libhackrf version: 2021.03.1 (0.6)
Found HackRF
Index: 0
Serial number: xxxxxxxxxxxx
Board ID Number: 2 (HackRF One)
Firmware Version: 2022.09.1 (API:1.06)
Part ID Number: xxxxxxxxxxx
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff

Finally,

Downgraded both host/firmware software and to my surprise.. same behavior. End conclusion, I'll try on another fresh setup and see if the behavior is the same. Good news, is it works, it's just maybe this delay was there all along and I just never paid much attention. Sorry for possibly a false alarm or some strange behavior on the computer I just happened to have at the moment to test on. I'll come back with some other results.

alphafox02, 2022-09-29:
I meant to say, subsequent runs of hackrf_info after that initial 10-15 seconds has no delay and does not need sudo. Meant to include that bit of info.

martinling, 2022-09-29:
It sounds like your system is very slow to apply udev rules after a device is connected. I'm not sure how that would happen. Perhaps you have something strange in some other file in `/etc/udev/rules.d` that's slowing it down?

alphafox02, 2022-09-29:
@miek I thought maybe that was it as udev was being spammed by tun0 (openvpn server/mini net/etc). I killed all that and observed plugging in the hackrf, seemed as though output was immediate in specific to the usb device (hackrf), however the delay remained. I'll go with another computer and try again. I think it's probably nothing with the project and more so a small issue on my end.

`udevadm monitor 
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[18400.427427] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3 (usb)
KERNEL[18400.429599] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0 (usb)
KERNEL[18400.429846] bind     /devices/pci0000:00/0000:00:14.0/usb1/1-3 (usb)
KERNEL[18405.833696] change   /devices/pci0000:00/0000:00:01.0/0000:01:00.0/drm/card1 (drm)
UDEV  [18405.845043] change   /devices/pci0000:00/0000:00:01.0/0000:01:00.0/drm/card1 (drm)
UDEV  [18411.758641] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3 (usb)
UDEV  [18413.163991] add      /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0 (usb)`

alphafox02, 2022-09-29:
Closing this out as I got around to another computer. Zero issue at all, new libs/firmware and hackrf_info works instantly without sudo. I'll chalk it up to something on the computer/setup that I have done lots of other experiments on and done who knows what to.

closed 2022-09-29 by alphafox02