greatscottgadgets/hackrf #1196: [Bug Report]: Operacake Time Mode

aoustas, 2022-10-07:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

Hello guys. I've just received my Operacake and I started experimenting straight away. All works as expected, Hackrf (latest fw installed) can see it and I can manually set the port route with the appropriate LEDs lighting up. When I tried using the time mode as per HACKRF documentation page, the A-B ports default to the last known route and they stay fixed. There is no switching happening whatsoever. I've tried both commands (hackrf_operacake -m time -t A1:1000 -t A2:1000 -t A3:1000 -t A4:1000  &  hackrf_operacake --mode time -w 1000 -t A1 -t A2 -t A3 -t A4) but the outcome is the same. Any ideas on how to solve this ?

### What are the steps to reproduce this?

Attached Operacake to HACKRF and issue the time mode command.

mossmann, 2022-10-07:
The `hackrf_operacake` command configures time mode, but actual switching in time mode only takes place during receive or transmit. Have you tried a receive or transmit operation after configuring time mode?

aoustas, 2022-10-07:
Yes I did with both GNU Radio as well as your examples for the DF. Still no switching. Is there a specific setting I need to check or configure ?

mossmann, 2022-10-07:
Do you have a PortaPack installed along with the Opera Cake?

Please provide the output of the commands `hackrf_info` and `hackrf_operacake -l`.

aoustas, 2022-10-08:
Hello again. No I don't have a PortaPack. I actually managed to solve the issue this morning. I was starting the tx/rx operation from a different terminal! As soon as I configured Operacake and started the operation from the same terminal all worked out. Thank you again. 

On a different note, is there a way to configure Operacake to work in 8:1 mode ?

mossmann, 2022-10-08:
8:1 operation is not a separate mode. You can configure port A0 to connect to any A port or B port (except B0) in any mode.

It is strange that running a command in a different terminal made a difference. I don't think that should happen. Are you able to reproduce the failure?

aoustas, 2022-10-08:
What I mean is to have A0 port to cycle through A1-A4 and continue to B1-B4 in time mode.

Yes I can reproduce it any time. Maybe it's something to do with the version of ubuntu I'm using or a different configuration on my test pc. I haven't tried to reproduce it on windows yet.

gozu42, 2022-10-19:
repro steps:

```
[user@dev-gsg gnss]$ hackrf_operacake -m time -w 1000000 -t A1 -t A2 -t A3 -t A4 -t B1 -t B2 -t B3 -t B4
(idle led pattern is A1-B1)
[user@dev-gsg gnss]$ hackrf_sweep -1 > /dev/null
(blinking as expected)

[user@dev-gsg gnss]$ hackrf_operacake -g
GPIO test passed
[user@dev-gsg gnss]$ hackrf_operacake -m time -w 1000000 -t A1 -t A2 -t A3 -t A4 -t B1 -t B2 -t B3 -t B4
(idle led pattern is A4-B4)
[user@dev-gsg gnss]$ hackrf_sweep -1 > /dev/null 
(no blinking)
```

aoustas, 2022-10-27:
Hello. Sorry for the late response. Although I see somebody already replied with a possible explanation of the problem, please find attached the output of the commands. 
[image]

straithe, 2022-12-05:
@aoustas do you still need assistance, or may I close this issue?

aoustas, 2022-12-05:
since i found a workaround i good for now you may close this issue. thanks.

closed 2022-12-05 by straithe