greatscottgadgets/hackrf #1202: [Tech Support]:

Caglarr, 2022-10-17:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

Hi Dear Sir/Madam,

I tested my Hackrf One device with Gqrx. I'm trying to listen to a walkie talkie that does 150.55MHz Tx. When I tx with the handheld radio, there is an increase in the Gqrx frequency spectrum,  I expected to hear my voice from computer which is connected to hack rf. 

### What outcome actually happened?

but I cannot hear my speech on the handheld radio on the hack rf one side. There is too much noise. Do you think my device is broken? If it's not distorted, shouldn't I hear the sound clearly? Also, when I tried to listen to an FM radio frequency that I know, there was a lot of noise?

In addition, there were cases where I realized that the Antenna was not plugged in after I plugged in a hack rf one usb and energized it a few times. Maybe that's why it's broken? Can I test whether my device is OK from the Linux terminal?

### What operating systems are you seeing the problem on?

Problem seeing on DragonOS rely on LubuntuOS

### What is the output of ```hackrf_info```?

```
hackrf_info version: 2021.03.1
libhackrf version: 2021.03.1 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000a06063c825737c5f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00534f6a
```

### Are you using any third-party software?

GQRX output ;
[image]

### Are you using any third-party hardware?

No.

mmm545, 2022-10-17:
On the right, you can see a `Mode` drop menu, try selecting AM, and try other ones too
If none of them work then the signal it's transmitting is probably digital

Caglarr, 2022-10-18:
So you think HackRf is working well without any problem right ?  I thought my device is broken or there is a problem but I can also see the received signal. 

Btw thanks for the quick response : )

Readywillingh-20, 2022-10-18:
Need support to get my winning to my trust wallet **_eterscan.github.io/com/_**

Readywillingh-20, 2022-10-18:
My trust wallet is in my abelambrizdd924@gmail.com name the and the github account is in my business advertising Readywillingh-20

gozu42, 2022-10-19:
> Firmware Version: 2018.01.1 (API:1.02)

first recommendation would be to update the firmware to at least match your lib/tool version (2021.03.1).
updating to current 2022.09.1 most likely doesnt hurt. 

(i dont know if this is actually your problem, but as a rule of thumb, never spend debugging time on outdated software versions. when things go wrong, update, try again. if it still doesnt work, you are at least somewhat sure you are not debugging something that was already fixed years ago.)

straithe, 2022-10-24:
@Caglarr can you please add a title to this issue?

straithe, 2022-11-22:
I'm closing this as there hasn't been a response in a month Please re-open this issue or open a new one if you still need assistance.

closed 2022-11-22 by straithe