greatscottgadgets/hackrf #1207: [Tech Support]:

konikofi, 2022-10-24:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

 Not a huge harmonic spike at 250MHz up from the desired frequency of 5300MHz.
[image]

### What outcome actually happened?

Seeing what appear to be harmonic spikes around +/- 250MHz, not sure if it is related to IF-gain or DC offset. Lowering IF-gain or disabling main gain does not appear to help.

### What operating systems are you seeing the problem on?

Fedora 36 with kernel 5.19.11. Trying to emulate radar pulses for wifi DFS testing.
[image]

### What is the output of ```hackrf_info```?

hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000087c867dc2b63b25f
Board ID Number: 2 (HackRF One)
Firmware Version: 2022.09.1 (API:1.06)
Part ID Number: 0xa000cb3c 0x00714758
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff
Operacake found, address: 0xff

### Are you using any third-party software?

gnuradio-companion
3.10.1.0 (Python 3.10.7)

### Are you using any third-party hardware?

sma output directly cabled to a signal hound bb60c spectrum analyzer

straithe, 2022-10-24:
Can you please add a title to this issue?

mossmann, 2022-11-07:
The spur at 5.56 GHz is the second harmonic of the IF which is normal. The confusing thing (to me at first) was why this spur appears to be at much higher power than the intended signal at 5.3 GHz, but I now realize that this is because the power at 5.3 GHz is being underreported.

One reason for this is that your flowgraph's output samples are very sparse. The IF harmonic is active 100% of the time, but the intended signal is only active 1/1429 of the time.

Another reason for this is the configuration and/or capabilities of your spectrum analyzer. It appears to be effectively averaging the power over time, so the sparse intended signal is represented at a much lower power than its true peak power.

To get an idea of the peak power, you could transmit constant samples instead of sparse samples. When I do this, I see output power at 5.3 GHz higher than the power of the spur I get at 5.56 GHz. If I then make the output sparse, the characteristics of the intended signal change, but the IF harmonic does not change.

mossmann, 2022-11-07:
Another thing that may be unintentional in your flowgraph is that the Osmocom Sink is configured with bandwidth that is 10 times the sample rate of 2 Msps. This results in several aliases over a the 20 MHz bandwidth and is the reason for the gently sloping sides of the peak at 5.3 GHz in your spectrum analyzer plot.

If you are trying to approximate an impulse, I recommend using a sample rate of 20e6 rather than 2e6.

closed 2023-01-05 by straithe