greatscottgadgets/hackrf #1213: [Documentation]; Information on hackrf kernel module

bm16ton, 2022-10-27:
### What would you like us to add to or change about the HackRF One documentation?

I know its old and mostly unused but im interested in the linux kernel module. Unfortunatly all google searches bring up page after page of the exact same quote about how it needs to be unloaded. Im curoous how to use it, what its capabilities where/are and what userland apps if any could interact with it. Im sure docs exist but are simply flooded out in my searches. Thank you guys for all the amazing work.

martinling, 2022-10-28:
The kernel driver was not written by us and we don't support or recommend it.

It is of very little use in practice. It was apparently part of an effort to support SDR devices directly in the kernel and expose them via an extension of the Video4Linux2 API. That API is documented here: https://www.kernel.org/doc/html/v4.9/media/uapi/v4l/dev-sdr.html

However, to the best of our knowledge none of the major SDR software ever added support for accessing devices using this API.

There is no real motivation for software to use it, since the existing user-space APIs through libhackrf and libusb (and similar projects for other SDRs) already work, have good performance, and are usable cross-platform - whereas V4L2 is Linux-specific.

bm16ton, 2022-10-28:
Thank ypu so much for your response! i realy do appreciate it i for sum
reason seldom get replies again thank you. Im just learning how to program
and love programming in the linux kernel. So this is more a
educational/labor of love situation, logic and reason dont apply (and
seldom do for me ;) I guess it makes sense because especially at the time
thats the place where things with tuners and the like would go, so closest
fit for the shoe horn. Thank you again and if uu guys ever add usb support
for libopencm3 and lpc43xx or a compatible general purpose usb driver that
integrates well will lopcm3 please let me know! I mnaaged to do a little
bit between greatfet and hackrf usb stacks on my lpc-link 4370 but when
your as dumb as me its hard going. Oh and love your products thank you for
them as well kind sir.

straithe, 2022-11-01:
Thank you. I'm going to close this issue as complete.

closed 2022-11-01 by straithe