greatscottgadgets/hackrf #1214: [Question]: Operacake switching while sweeping

bennymeg, 2022-10-29:
### What would you like to know?

Hi,

I am trying to switch a port in Operacake after each sweep [0-6GHz] is completed.

I have updated the `hackrf_sweep.c` code such that after _sweep_count++_ [(line 267)](https://github.com/greatscottgadgets/hackrf/blob/0afc9857e4f08c2200c9f4412e8816dbacd9e36b/host/hackrf-tools/src/hackrf_sweep.c#L267) is executed, I issue a call to _hackrf_set_operacake_ports()_ which unfortunately doesn't work, as I get an error code saying the resource is busy (-1000). Operacake is set to Manual mode (again, programmatically).

I know that switching ports while sweeping is possible, because it works in Frequency mode.

Digging a bit deeper into the firmware `(usb_api_operacake.c)`, I see that _usb_vendor_request_operacake_set_ports()_ doesn't have a handler for USB_TRANSFER_STAGE_DATA, but only a handler for USB_TRANSFER_STAGE_SETUP (unlike in the case with frequency and dwell modes), so I must assume the functionality I am after is unsupported at this moment. Am I correct? and if so, is there any way to work around this current limitation?

Thanks in advance,
Benny.

martinling, 2022-10-29:
> I have updated the hackrf_sweep.c code such that after sweep_count++ [(line [...]

The reason that you get a "resource busy" error is that `hackrf_set_operacake_ports` calls `libusb_control_transfer`, which is [documented](https://libusb.sourceforge.io/api-1.0/group__libusb__syncio.html#gadb11f7a761bd12fc77a07f4568d56f38) to return `LIBUSB_ERROR_BUSY` if it's called from an event handling context.

You're in an event handling context because you're making the call from the RX callback in `hackrf_sweep`, which gets called as part of the libusb event handling when a USB transfer completes.

It's not possible to use the libusb synchronous API from within an asynchronous event handler. You could make this work by having the RX callback just set a flag, and then having the main loop call `hackrf_set_operacake_ports` when that flag is set. Alternatively,  from the RX callback, you could use `libusb_submit_transfer` to submit the same request through the asynchronous API.

Either way, requesting the change from the host is going to be somewhat inaccurate in terms of timing. By the time you have the RX callback with the last data from one sweep, the device has already started the next sweep.

If you want to be sure the Opera Cake switches at the change from one sweep to the next, it would be better to implement this on the firmware side, by inserting the necessary changes in the `sweep_mode` loop in `usb_api_sweep.c`.

> Digging a bit deeper into the firmware (usb_api_operacake.c), I see that usb_vendor_request_operacake_set_ports() doesn't have a handler for USB_TRANSFER_STAGE_DATA, but only a handler for [...]

No, this difference does not mean those functions are unfinished. The handling for `set_ports` does not need a DATA stage, since all the information needed for the request is encoded in the SETUP packet. For `set_ranges` and `set_dwell`, there is additional data needed, so those requests use a SETUP packet followed by a DATA stage.

bennymeg, 2022-10-29:
Thank you for the quick response!

straithe, 2022-11-01:
@bennymeg it looks like your question has been answered, may I close this issue?

bennymeg, 2022-11-01:
Yes, I'll close it now.
Thanks.

closed 2022-11-01 by bennymeg