greatscottgadgets/hackrf #1216: [Tech Support]:

BenHut1, 2022-10-31:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

I was hoping to download the compiled binary of the latest version of HackRF Tools, so I in the latest release I clicked the "hackrf-2022.09.1.zip" link, making sure to NOT click the "Source code (zip)" link.

### What outcome actually happened?

When I clicked the link that appeared to be the correct one for the compiled binary, the file that was downloaded was in fact NOT the compiled binaries of the software. Instead, it just contained the source code. I then (thinking that maybe you accidentally switched the files) also tried downloading the zip file called Source Code, but it just had more source code files in it. Neither of these zip files contained the compiled binaries.

### What operating systems are you seeing the problem on?

not applicable to the problem in question

### What is the output of ```hackrf_info```?

not applicable to the problem in question

### Are you using any third-party software?

not applicable to the problem in question

### Are you using any third-party hardware?

not applicable to the problem in question

mmm545, 2022-10-31:
your other option is [compiling it yourself](https://github.com/greatscottgadgets/hackrf/tree/master/host)

BenHut1, 2022-10-31:
If I follow the Ubuntu instructions for compiling hack rf tools, will the
result only be runable in Linux? Or will it also perform cross-compiling,
and create a version that will run in Windows?

mmm545, 2022-10-31:
no, if you want to build for windows, you have to follow the building instructions for windows, if you want to build for linux, you have to follow the instructions for linux etc

straithe, 2022-11-01:
@BenHut1 can you please add a title to this issue?

BenHut1, 2022-11-05:
> windows binaries [HackRF-Windows-Tools.zip](https://github.com/greatscottgadgets/hackrf/files/9902023/HackRF-Windows-Tools.zip)

Thanks for that.

BenHut1, 2022-11-05:
> @BenHut1 can you please add a title to this issue?

That's strange. I thought I added a title when I wrote this issue.

straithe, 2022-11-07:
@BenHut1 it looks like your issue has been addressed. May I close this issue?

sdb333777, 2022-11-07:
i was just trying to help out that other guy, you can delete this if you want.

BenHut1, 2022-11-07:
> 

That's fine.

closed 2022-11-07 by straithe