greatscottgadgets/hackrf #1218: Not seeing HackRF on Windows [Tech Support]:

nikstp, 2022-11-01:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

I'm not completely sure how to do a basic test on Windows, but I've tried running "hackrf_info" in both CMD and powershell. The HackRF does come on as the LEDs light up when it's plugged in and when I put it into the DFU mode, it pops up as a device on my computer. 

### What outcome actually happened?

I get "'hackrf_info' is not recognized as an internal or external command, operable program or batch file."

### What operating systems are you seeing the problem on?

Windows 10

### What is the output of ```hackrf_info```?

N/A

### Are you using any third-party software?

Need to use it with GNU Radio. 

### Are you using any third-party hardware?

N/A

straithe, 2022-11-01:
How did you install the HackRF software on your system?

nikstp, 2022-11-01:
@straithe Ah, nevermind. I installed the Pothos SDR binaries, but I did not follow thru with proper installation of the USB driver with Zadig. Also, for Windows, Pothos SDR creates an executable "hackrf_info" which is placed in PothosSDR/bin, so I ran that and it gave me some info back:

hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f77c60dc253f56c3
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x00544f5a

Seems like this can be closed out. Thank you!

closed 2022-11-01 by nikstp