greatscottgadgets/hackrf #1221: [Tech Support]: HackRf not being recognized by USB

jozamm, 2022-11-07:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

Correct Operation of HackRF

### What outcome actually happened?

I charged the battery and by mistake plugged the port into Quick charge port 3.0.  It seems that the battery charged correctly than all lights switched off

After this event the HackRFone + PortPack H2 did not work. After removing the PortaPack and attaching the HackRF to the Laptop alone, only the 3.3V, RF and 1.8V LED's came on and no USB connectivity. Under Linux I could not open the device. When trying to use the DFU method the 3.3V and RF LED's light up with no USB connection. Previously it worked with the laptop so it not a port and cable issue.

I ran some other tests as follows and I totally baffled

Tests run

DC Voltage on 1.8V and 3.3V correct - Checked by multimeter
Current Consumption 170mA, DFU mode 70mA (Approx)
The LPC oscillator is running at 25MHz - Checked by Oscilloscope at XTAL pins
I checked the W25Q80, the LPC is accessing it as I saw activity using an oscilloscope on the CLK, DI and DO lines
There is no activity on the D+ and D- USB Lines

I ran some other activity tests for example I2S0_TX_SCK and there is a signal on it.

So the LPC seems to be running

### What operating systems are you seeing the problem on?

Windows 10
Ubuntu Linux 20.04 and 22.10

Both in normal and DFU mode the board is not detected in Linux, tried on Linux 22.10 latest hackrf package

### What is the output of ```hackrf_info```?

HackRf said that No Boards Detected

### Are you using any third-party software?

No

### Are you using any third-party hardware?

Board was attached PortaPack H2 when it stopped working

straithe, 2022-11-07:
In your description it says that you have tried the board on Windows and Linux, but that the board is not detected on Linux. Is the board detected on Windows?

jozamm, 2022-11-07:
Hi,

On windows it is not detected neither in normal mode nor DFU mode.

However I noted that the LPC went into DFU mode because when I switched to DFU the W25Q80 was not accessed by the LPV

straithe, 2022-11-07:
The outcome you have listed is "correct operation of HackRF One". Could you please be more detailed? What do you want to be able to do with the HackRF?

jozamm, 2022-11-07:
Hi, I need to get HackRF to establish a USB connection to the PC. That way I would confirm that it works. 

After connect it to PortaPack H2 again

Thanks

straithe, 2022-11-07:
Can you please add a picture of your HackRF (not in the Portapack) to this issue?

jozamm, 2022-11-07:
Hi I, Here is a photo

[image]

straithe, 2022-11-07:
This is not a Great Scott Gadgets HackRF. It sounds like you do have a hardware issue. Please contact the person or company that you purchased your HackRF from to see if they can offer you a replacement.

closed 2022-11-07 by straithe