greatscottgadgets/hackrf #1222: [Question]: What is the return value from the callbacks supposed to be?

BenHut1, 2022-11-07:
### What would you like to know?

When writing a TX or RX callback function for the HackRF, what is the return value from that function supposed to be? And how does the HackRF handle different return values from this callback function?

martinling, 2022-11-07:
The callback should return zero if it wants to be called again for more data. If it returns anything other than zero, the TX/RX streaming comes to an end and it won't be called again.

BenHut1, 2022-11-07:
> The callback should return zero if it wants to be called again for more data. If it returns anything other than zero, the TX/RX streaming comes to an end and it won't be called again.

Thanks for that info. I wish that was in the official documentation. I couldn't find that mentioned in the PDF doc, or even in the hackrf.h file in the source code.

straithe, 2022-11-07:
Adding this information to the documentation is on my very long list of things to do. I've added the documentation tag to this issue and will leave it open until I've done the documentation update containing this information.

BenHut1, 2022-11-09:
> The callback should return zero if it wants to be called again for more data. If it returns anything other than zero, the TX/RX streaming comes to an end and it won't be called again.

I noticed something interesting. While it doesn't get called again, it doesn't completely stop streaming. Specifically, it doesn't shut off the RX or TX LED on the HackRF one. Possibly on TX, transmission is continuing using the last supplied buffer (though it can't get another buffer because the callback never gets called again), though I haven't yet tested if the transmission continues in this situation. Previously, I had assumed that the callback returning a value other than 0 would shut off the RX or TX activity completely, based on your description, but it doesn't, based on the tests I've run. I discovered I still need to explicitly call hackrf_stop_tx or hackrf_stop_rx to stop TX or RX completely. Only after calling one of those functions does the RX or TX LED turn off. Is this a bug, or is this intended behavior?

martinling, 2022-11-09:
Yes, you still need to call `hackrf_stop_tx` or `hackrf_stop_rx`, because those functions send the USB request to the device which tells it to change modes. Until that request is sent, the device doesn't know that the host isn't going to send more TX data or request more RX data, even though the current libhackrf API doesn't give you a way to do so.

In principle, libhackrf could have been made to send a stop request as soon as the TX/RX callback returns nonzero. But the library can't just call into `hackrf_stop_{tx|rx}` for you at that point, because those functions use the synchronous libusb API, and the context which the callback is executed from is an event handler within the asynchronous libusb API. The stop request would need to be instead submitted as an asynchronous transfer, and then the background thread doing the event handling would need to continue until that transfer completed. If an error then occurred during that transfer, the background thread would have to signal that error back to the main thread somehow so it could decide what to do next.

In practice, it works out a lot simpler to just let the main thread make the call to `hackrf_stop_{tx|rx}` itself, and then it's in control of what to do if any error happens.

The latest release, 2022.09.1, includes several changes to do with stopping TX/RX that you may also want to know about:

- If you don't stop TX, the device won't retransmit previous buffer contents. The firmware tracks the available data in the buffer and will send zeroes to the DAC whenever the buffer is empty. So in the situation you observe where `hackrf_stop_tx` has not been called, the TX light is still on and the RF path is still powered up, but its inputs are zero.

- To ensure that you don't call `hackrf_stop_tx` too early, before the device has finished transmitting the last data you sent from the TX callback, you can call `hackrf_enable_tx_flush` before you call `hackrf_start_tx`. You pass it a callback, which will be called when the last data is guaranteed to have been transmitted. That callback can then signal your main thread by whatever means you prefer, so it knows it's now time to call `hackrf_stop_tx`.

- If you want to ensure that the device leaves TX mode even if your program crashes, computer crashes, USB fails etc then you can call `hackrf_set_tx_underrun_limit` to set a sample count after which the device will automatically return to idle mode if the TX buffer has stayed empty for that long.

BenHut1, 2022-11-09:
> 
> In principle, libhackrf could have been made to send a stop request as soon as the TX/RX callback returns nonzero. But the library can't just call into `hackrf_stop_{tx|rx}` for you at that point, because those functions use the synchronous libusb API, 

The hackef_stop functions don't seem to be completely synchronous. If you try to switch from rx to tx by calling the stop rx function and then immediately call the start tx function, it doesn't start the tx mode. It seems that stop rx function returns before the hackrf has actually switched back to idle state. So it's not completely synchronous. I found between my call to the stop function and my next call to the start function, I needed to repeatedly call the hackrf_is_streaming function in a loop, until it returns a value other than 1. That's the only way to be sure that the streaming has stopped.

.
> The latest release, 2022.09.1, includes several changes to do with stopping TX/RX that you may also want to know about:
>
>     * If you don't stop TX, the device won't retransmit previous buffer [...]

So did older versions of hackrf.dll simply repeat the last provided buffer?

martinling, 2022-11-09:
> If you try to switch from rx to tx by calling the stop rx function and then immediately call the start tx function, it doesn't start the tx mode. It seems that stop rx function returns before the [...]

This should not happen with the 2022.09.1 release, which includes a lot of fixes to start/stop behaviour which I worked on over the past year. I would not be surprised if it happens with older versions. If you are seeing this on the 2022.09.1 release of libhackrf, please open a bug report.

> So did older versions of hackrf.dll simply repeat the last provided buffer?

It's not the version of the DLL that affects this but the version of the firmware. Older versions would repeat the last buffer contents if the host stopped sending new data. This was bug #180 and was fixed by PR #982, which is included in the 2022.09.1 release.

BenHut1, 2022-11-09:
> 
> It's not the version of the DLL that affects this but the version of the firmware. Older versions would repeat the last buffer contents if the host stopped sending new data. This was bug #180 and was fixed by PR #982, which is included in the 2022.09.1 release.

I'm tempted to upgrade to the latest firmware, but will that break its compatibility with my currently used hackrf.dll? I'm using the 32bit version of the dll that I compiled myself (I couldn't find a 32 bit version anywhere on the internet) as the program I wrote that uses the dll is 32bit, but it was a major PAIN to compile that dll, (compiling the hackrd library is not an easy or simple task at all). So I hope to not need to compile the dll again if I upgrade the firmware.

martinling, 2022-11-09:
> I'm tempted to upgrade to the latest firmware, but will that break its compatibility with my currently used hackrf.dll?

No, the firmware and libhackrf versions don't have to match, you can use the newer firmware with your older libhackrf, you just won't have access to some new features in the firmware that your older libhackrf doesn't know about. However, there were quite a few fixes in libhackrf too, so I'd recommend upgrading if you can.

If it's any help, I wrote up some instructions recently for building with MSVC on Windows using dependencies installed with `vcpkg`: https://gist.github.com/martinling/f0acfa5af67777ed4345471a29a2bf3a

I haven't verified this, but I think that if you follow those instructions replacing all instances of `x64`  with `x86`, that will get you a 32-bit build.

dmaltsiniotis, 2022-11-09:
To add to @martinling's instructions, here are my build notes (in markdown format) for building the host tools with modern versions of Visual Studio community edition: https://gist.github.com/dmaltsiniotis/bb761a9b9a642e9407f14c6f7dfc466a I've been waiting to clean them up a bit before publishing, but now's as good as time as any. These instructions assume you've already acquired the three other dependencies (pthreads, fftw, etc...) and honestly that's the hardest part of building on Windows really. It's pretty straightforward once the dependencies are there.

straithe, 2022-12-05:
@BenHut1 do you still need assistance with this issue?

closed 2023-04-23 by straithe