greatscottgadgets/hackrf #1224: [Question]: What path does the device get mounted as?

jrj2211, 2022-11-11:
### What would you like to know?

I'm looking to use the HackRF inside a Debian docker container. To do this I need to manually specify the device to pass it through to the container:

```
docker run -it -device=/dev/ttyUSB0
```

Issue is, I cant figure out what path the device gets mounted as. I have no ttyUSB0, which I was expecting since its a USB device. Looking through all the files in my /dev folder, none look like it and if I unplug the device, nothing is removed from that list.

[image]

Does anyone know where the path to the device would be? I'm running on a Raspberry PI with the latest RaspiLite OS.

straithe, 2022-11-11:
Containers and VMs are notorious for not doing USB hand-off well. I suggest trying to use your HackRF with an OS that is not containerized.

martinling, 2022-11-11:
It wouldn't be `/dev/ttyUSB0`, that would be for a USB serial port only.

The device node you actually need will be a generic one under `/dev/bus/usb` somewhere.

You can see the Docker arguments we use on our hardware CI server here: https://github.com/greatscottgadgets/hackrf/blob/15636efb4d05948991fd193efcd41e493eb3bb9a/Jenkinsfile#L3-L5

Group ID 46 is usually the `plugdev` user on Debian-based systems, but you may need a different value if that differs on your system.

jrj2211, 2022-11-11:
@martinling Thanks for the info! I figured it was showing up somewhere in one of the many folders in /dev but didn't know where to look. I'll give this a shot.

straithe, 2022-11-28:
@jrj2211 it looks like your question has been answered. May I close this issue?

closed 2023-01-05 by straithe