greatscottgadgets/hackrf #1225: [Question]:

alexmealster19, 2022-11-14:
### What would you like to know?

I want to communicate with my hackRF through serial but I'm not sure exactly where to start. My google searches haven't turned up much so if someone could guide me to the right place I would very much appreciate it.

My goal is to power, send and receive data from the hackRF from my raspberry pi 4.

dmaltsiniotis, 2022-11-14:
Hi @alexmealster19, The HackRF One communicates via USB normally/natively, which is supported on a RPI4. Can you elaborate on your use-case requirement for serial communication (or in general what you're trying to do)? If you use the native libraries and and tools, you can power, send, and receive data on a RPI4 pretty easily over USB.

alexmealster19, 2022-11-14:
Hi @dmaltsiniotis, thanks for your response! I have been able to establish successful communications with the hackRF over usb. I want to use serial because I plan to fly the hackrf as a payload on a small satellite for a university project and usb is a non standard connection type for flight projects. If I were able to communicate via serial I could establish a much more secure connection to the hackrf.

martinling, 2022-11-14:
Well, there is a UART on the LPC4320 broken out to pins 19 and 20 on the P22 header. But you'd have to modify the firmware to make it do anything. And you wouldn't be able to go faster than a few Mbps at most over it, which is less than a hundredth of the bitrate of USB, so you'd not be able to get much data in and out.

If you're serious about building a flight payload based on a RPi4 and a HackRF, you'd probably have to redesign both devices completely anyway to fit into the cubesat form factor and pass all the required testing, so this is really the least of your issues.

straithe, 2022-11-28:
@alexmealster19 it looks like your question has been answered. May I close this issue?

closed 2022-11-28 by alexmealster19