greatscottgadgets/hackrf #1226: [Question]: Looping hackrf_sweep x times

alexmfz, 2022-11-14:
### What would you like to know?

Hello,
The question I would like to ask You is: Is it possible to run hackrf_sweep in OneShot mide x times with a timer?

I have done it but when i tried to run them, the script stops and the output doesnt get any result.

martinling, 2022-11-16:
Are you running the latest version? Check the output of `hackrf_info`.

The recent 2022.09.1 release fixed a lot of issues in both the firmware and host software that could cause stopping and restarting the HackRF to fail.

Previously there were several bugs in this area (e.g. #883, #916). All these issues were fixed in PRs #1029 and #1042. Those fixes are included in the 2022.09.1 release. If you are running an older version of either the firmware or libhackrf you should upgrade.

alexmfz, 2022-11-16:
Hi, 
I have seen that the output of hackRF_info is:
[image]

It seems is not upgraded. How could I do it?

Thank you guys :)

martinling, 2022-11-16:
You can download the new release here: https://github.com/greatscottgadgets/hackrf/releases/tag/v2022.09.1

Then follow the instructions in the documentation for [updating firmware](https://hackrf.readthedocs.io/en/latest/updating_firmware.html#updating-the-spi-flash-firmware) and [installing the host software from source](https://hackrf.readthedocs.io/en/latest/installing_hackrf_software.html#installing-from-source).

alexmfz, 2022-11-21:
Okey, 
I will upload it as you said. 
Thank you so much for the info, I will let you know the results :)

alexmfz, 2022-11-21:
Okey, 
I will upload it as you said. 
Thank you so much for the info, I will keep you inf

alexmfz, 2022-11-21:
Hi again guys,

I have just upload to the last version: 
[image]
But Still I am having null outputs
[image]

alexmfz, 2022-11-21:
Hi @martinling ,
In response to your comments:
1. Respect from the first one, with **endConnection()**, I meant **hackrf_close() and hackrf_exit()** functions (like in the original code, I just regroup them  into that function). And It is called from  my main thread.
Here is a screenshot where the function **reconfigureHackRF()** is called (into the main) and the functions called in **reconfigureHackRF()**.

As you could note, in the screenshot you can see what functions are called in **reconfigureHackRF()** (which are the ones in the black square). In the other hand, the loop shows a function call **hackRFTrigger()** which configures the timer and invokes **hackrf_start_rx_sweep()**

2. In response to your second comment, each iteration I close the hackRF with hackrf_close() and hackrf_exit(). But like you said I invoke everytime hackrf_init. 
So could be enough just to invoke in each iteration hackrf_init_sweep and hackrf_start_rx_sweep functions?

martinling, 2022-11-21:
> So could be enough just to invoke in each iteration hackrf_init_sweep and hackrf_start_rx_sweep functions?

Yes, that should be sufficient, if you don't need to close the device in between sweeps. The sequence you've been using should work too though, so I don't think that's the cause of your problems, although it would be interesting to know if you get different results from keeping the device open.

A few more questions if you don't mind:

- When the sweep fails, how are you detecting that? Where does the `success` value get set?

- Is `hackRFTrigger` is checking for errors from `hackrf_init_sweep` and `hackrf_start_rx_sweep`? Do you see any?

- Is your time between triggers definitely enough for the previous sweep to have finished properly?

alexmfz, 2022-11-22:
I dont need to close it. Just sweeping every 0.25s. 
I have tried it but I got the error  **hackrf_start_rx_sweep() failed: HackRF busy (-6)** at 2nd iteration. I will try it later again
About the questions you mentioned:

**1. When the sweep fails, how are you detecting that? Where does the success value get set?**  When the sweeps fails I detected it at the rx_callback function setting the variable to true if starts with the correct frequency . After the sweeping finished I set it to false to iterate again:
[image]

**2. Is hackRFTrigger checking for errors from hackrf_init_sweep and hackrf_start_rx_sweep? Do you see any?** Yes, the function  is checking for errors for hackrf_start_rx_sweep as you can see in this screenshot.

[image]

hackrf_init_sweep function is checked in a previous function.

**3. Is your time between triggers definitely enough for the previous sweep to have finished properly?** Yes, I have calculated the time and the sweeping time is under 0.25s. For 45-245MHz is around 0.05 s. (The timer handler set the flags to 1 to exit from the while of the hackRFTrigger function

martinling, 2022-11-22:
> I have tried it but I got the error hackrf_start_rx_sweep() failed: HackRF busy (-6)

I think that was because you weren't calling `hackrf_stop_rx()`. When your RX callback sees the last data you care about and you return a non-zero value from the callback, libhackrf stops requesting data from the device and you stop getting callbacks. But the HackRF is still in sweep mode, and still running further sweeps, even though you're not asking for the data any more. Your main thread needs to call `hackrf_stop_rx()` to stop it, so your callback has to somehow let the main thread know when it's time to do that.

So when you call `hackrf_start_rx_sweep()` without having called `hackrf_stop_rx()`, you get the busy error because it's already sweeping.

This isn't very well documented at the moment, which is on our todo list to fix.

alexmfz, 2022-11-22:
Then the flux should be:
1. Configuration hackrf
2. Loop
    2.1. hackRFTrigger => Where I call hackrf_start_rx_sweep()
    2.2. hackrf_stop_rx()
    2.3. hackrf_init_sweep()
3. hackRF_close() 
[image]

Should this be correct?

martinling, 2022-11-22:
> Then the flux should be:
> 
>     1. Configuration hackrf
> 
>     2. Loop
>        2.1. hackRFTrigger => Where I call hackrf_start_rx_sweep()
>        2.2. hackrf_stop_rx()
>        2.3. hackrf_init_sweep()
> 
>     3. hackRF_close()
> 
> Should this be correct?

I think you want this order in the loop:

2. Loop
    2.1. hackrf_init_sweep()
    2.2. hackrf_start_rx_sweep()
    2.3. Wait for sweep to complete (use some event set in your callback before returning nonzero).
    2.3. hackrf_stop_rx()

alexmfz, 2022-11-23:
Hi @martinling,

I think I have discover an easier way to do it.
Instead of triggering the call back function in one shot mode, I am able to run it in infinite mode and trigger the capture data between 0.25s.
After some test, I saw that this way of thinking works corretly :)

martinling, 2022-11-23:
Yes that should also work. I thought perhaps you were using one-shot sweeps because you needed to sweep different frequency ranges, but if it's the same range each time then that approach may indeed be simpler.

straithe, 2022-12-05:
@alexmfz it looks as though you may have resolved your issue. May I close this GitHub issue?

alexmfz, 2022-12-15:
Yes It can be closed.

Thank you si much for your help :)

closed 2022-12-15 by alexmfz