greatscottgadgets/hackrf #1229: HELP ! Broken hardware component.. How to fix it ?

HaroGaro1312, 2022-11-16:
### What would you like to know?

  Hi, 

  A few weeks ago I’ve bought a HackRF One  (Great Scott Gadgets) and a Portapack H2+ (clone), when I tried to mount the Portapack onto the HackRF One unfortunately the battery pushed onto a component and finished to broke the component.. (see the pictures) 
[image]
the Portapack was initially hard to fit on, so I didn’t noticed when it reached the component.. I left that on side for a few weeks because I was disgusted of what happened.. But now I’m back.

  So, anyway, here’s my questions, does anyone have a solution for that ? Or an idea to fix it ? If it’s fixable of course.. Oh god.. Please tell me that’s fixable.. 

  I precise the HackRF One is still recognized on mac m1 with hackrf_info  

  The broken piece’s still hanging on but if I push a little bit under and on the side of it, the broken mark disappears (see the 2nd picture) but I think that doesn’t solve anything, am I wrong ? (I know that doesn’t solve anything but right now like the UFOs "I want to believe" that will work..) 
  
  Well if it doesn’t work (lol).. No seriously if you’ve got an idea of where I can get this component to replace the original broken one please tell me.. (See pictures) [image] And If I can repair it, is there is anything to do on the software/firmware or I just have to replace the component ? Or maybe if you know a repair shop where I can send the HackRF One for repair (if possible in France or Europe but even if it’s in US or New Zealand, I’ll take it). Honestly I’ll take any solution that can help me so if you have one please tell me, because I really need help.. 

  Well, a very big thanks in advance to anyone who come to help, seriously thank you very much if you can help on that problem. 

  Cordially, Anthony.

mmm545, 2022-11-16:
It's totally fixable! It's a 4.7μH inductor, the original part is NRG4026T4R7M, however other parts like the LQH44PN4R7MPRL is a good option.

martinling, 2022-11-16:
That part is a 4.7uH inductor which is part of the 5V to 3.3V DC-DC converter circuit. Physically, it's a coil of wire wound around a ferrite spool, and what you've cracked is just a corner of the flat plate at one end of the spool.

Honestly, missing that piece is unlikely to affect anything. If the coil were broken, the HackRF wouldn't function at all. If the broken corner will come away cleanly you can probably just remove it and leave it off. If it's still partially attached, it may be best not to pull on it. To keep the broken part in place you could drip a spot of liquid cyanoacrlyate glue into the crack and hold it shut till it dries.

If you do want to replace the part it's a straightforward soldering job. The part suggested above would be fine, another option is NRS4018T4R7MDGJ. Anything that's 4x4mm with the same inductance and similar current rating.

HaroGaro1312, 2022-11-17:
Hello

  Oooooh thanks !! I was so freaked out ! That was terrible, thanks very much for your fast answers both of you, it reassures me so much ! 

  The little part is still attached to the rest, so i push on the side (like on the 2nd picture) and I just have to put a little drop of superglue and that's it ? Cyanoacrylate is superglue right ? (Sorry my English isn't very good) 

I will just have a last problem, it's that the battery will still touch the inductor and impeach to put the HackRF with the Portapack in the case so I was wondering if maybe you've got an idea of which battery will be good for the HackRF to replace the original one which is a bit too thick ? I mean do you know which Ampere value is the battery ? (I don't have a multimeter around me) The one I've got have a little PCB attached to it and the values are 3.7V / 2500mAh (see the picture) [image]so have you got idea of a battery with the same value but less thick ? 

  Thanks again for your answer, I'm glad that seems not as severe as I was thinking at the beginning, and thank you by advance if you know the right battery to use if you don't that's not a problem, anyway thanks again for all you help, 

  Cordially, Anthony.

martinling, 2022-11-17:
Yes cyanoacrylate is the type called "super glue" or "crazy glue" in most places.

For the battery issues you'll need to take that up with the manufacturer of your PortaPack. The [original PortaPack design](https://www.sharebrained.com/) doesn't use an internal battery. It's something the clones have added.

HaroGaro1312, 2022-11-17:
[image]

  Well, it seems to be good, the cyanoacrylate have take in really fast so it was easy, I just had to pushed a bit on the side with a little screwdriver put a bit of superglue with a guitar cord and that was finished in 5min ! 

  Thank you very much for your answers if I had a problem with this part now I have the reference and a repair technic so thank you ! 

  If you know the answer to my precedent post, which type of battery choose to replace the original one which is too thick to fit in the case with the Portapack so if someone got an answer, I'll take it !  

  Well anyway, thank you again very much for all your help and answers, really thank you very much for all. 

  Cordially, Anthony.

mmm545, 2022-11-17:
You're welcome!
You may close this issue unless you need further help!

closed 2022-11-22 by straithe