greatscottgadgets/hackrf #1230: [Bug Report]: Inconsistent bin results from hackrf_sweep one-shot mode

grvvy, 2022-11-16:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

The target signal power of ~ -20dB should always appear in the second bin of this test output but it somewhat regularly appears shifted into a different bin or totally out of scope at either/both 9.5MHz and 2665.5MHz. Meanwhile 915.5MHz and 5995.5MHz are consistently rock solid.

Changing the sweep call to sweep twice instead of using one-shot mode results in the 1st sweep failing as does here, but the 2nd sweep never fails along with the 1st.

### What are the steps to reproduce this?

https://gist.github.com/grvvy/a1a9c970d05a6c8654bd1fd2aa579e34

Running this script with two hackrf's plugged in with antenna and clk ports connected.

### Can you provide any logs? (output, errors, etc.)

The test fails here after the 4th run (signal not appearing at 9.5MHz and signal (-20.73) shifted to the 9th bin at 2665.5MHz).
```
Running test...
2022-11-16, 15:49:23.103952, 9000000, 14000000, 333333.33, 60, -35.83, -23.87, -24.87, -39.90, -54.86, -71.27, -64.26, -67.41, -66.41, -72.92, -68.81, -65.44, -82.04, -66.58, -64.18
2022-11-16, 15:49:24.219668, 915000000, 920000000, 333333.33, 60, -28.07, -22.44, -28.43, -66.44, -74.80, -85.79, -77.05, -70.29, -64.99, -65.41, -65.00, -72.53, -76.89, -80.10, -80.51
2022-11-16, 15:49:25.336927, 2665000000, 2670000000, 333333.33, 60, -23.36, -17.69, -23.61, -57.42, -60.59, -63.11, -62.20, -64.62, -65.04, -69.21, -70.05, -69.54, -69.92, -86.45, -77.35
2022-11-16, 15:49:26.437785, 5999000000, 6004000000, 333333.33, 60, -26.17, -20.57, -27.03, -49.25, -41.32, -45.67, -49.85, -52.20, -51.93, -51.13, -59.96, -44.10, -42.67, -48.11, -46.92
Running test...
2022-11-16, 15:49:27.556318, 9000000, 14000000, 333333.33, 60, -28.21, -22.54, -28.54, -78.27, -80.53, -71.34, -71.99, -75.28, -98.66, -74.08, -72.13, -63.94, -63.68, -71.11, -83.52
2022-11-16, 15:49:28.672460, 915000000, 920000000, 333333.33, 60, -27.92, -22.49, -28.69, -65.33, -68.59, -71.14, -70.75, -71.24, -70.78, -72.50, -69.64, -66.08, -61.43, -61.76, -65.64
2022-11-16, 15:49:29.773170, 2665000000, 2670000000, 333333.33, 60, -27.96, -18.19, -20.58, -38.81, -53.63, -63.53, -63.89, -66.13, -72.82, -72.75, -72.81, -71.55, -83.98, -66.54, -67.53
2022-11-16, 15:49:30.874257, 5999000000, 6004000000, 333333.33, 60, -27.78, -20.88, -26.29, -48.29, -54.56, -46.73, -41.06, -45.51, -60.29, -55.09, -55.96, -52.79, -53.62, -45.81, -43.87
Running test...
2022-11-16, 15:49:31.976399, 9000000, 14000000, 333333.33, 60, -28.42, -22.56, -28.30, -64.18, -66.32, -69.76, -87.33, -67.56, -67.96, -82.00, -71.27, -66.06, -71.37, -83.48, -71.30
2022-11-16, 15:49:33.079739, 915000000, 920000000, 333333.33, 60, -28.64, -22.50, -28.03, -65.94, -64.44, -73.95, -76.47, -74.30, -83.27, -72.37, -70.76, -69.77, -76.12, -90.48, -70.31
2022-11-16, 15:49:34.206299, 2665000000, 2670000000, 333333.33, 60, -28.32, -18.29, -20.42, -38.24, -51.56, -62.19, -65.21, -68.79, -72.94, -73.62, -71.15, -81.35, -76.21, -75.81, -68.99
2022-11-16, 15:49:35.309317, 5999000000, 6004000000, 333333.33, 60, -25.76, -20.20, -25.72, -38.68, -44.96, -48.88, -44.07, -51.02, -52.54, -49.31, -54.38, -47.67, -44.90, -55.65, -44.75
Running test...
2022-11-16, 15:52:58.661475, 9000000, 14000000, 333333.33, 60, -68.91, -63.01, -65.03, -70.71, -69.87, -74.21, -78.06, -70.23, -65.43, -67.75, -64.02, -61.21, -65.44, -70.45, -81.64
2022-11-16, 15:52:59.780609, 915000000, 920000000, 333333.33, 60, -28.46, -22.44, -28.09, -60.67, -68.63, -67.65, -70.22, -69.83, -88.02, -70.50, -76.49, -67.40, -69.42, -79.39, -68.54
2022-11-16, 15:53:00.881921, 2665000000, 2670000000, 333333.33, 60, -56.13, -65.27, -52.08, -49.34, -41.03, -34.07, -25.91, -21.12, -20.73, -23.86, -28.60, -32.63, -37.10, -41.79, -46.88
2022-11-16, 15:53:02.006717, 5999000000, 6004000000, 333333.33, 60, -26.89, -20.80, -25.85, -44.79, -43.75, -48.94, -51.20, -46.01, -41.36, -46.54, -52.44, -46.46, -45.63, -53.35, -44.44
```

```
$ hackrf_info
hackrf_info version: git-e3e40649*
libhackrf version: git-e3e40649* (0.7)
Found HackRF
Index: 0
Serial number: 0000000000000000325866e629822923
Board ID Number: 2 (HackRF One)
Firmware Version: git-47b41567 (API:1.06)
Part ID Number: 0xa000cb3c 0x005f4754
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One

Found HackRF
Index: 1
Serial number: 0000000000000000325866e629a25623
Board ID Number: 2 (HackRF One)
Firmware Version: 2022.09.1 (API:1.06)
Part ID Number: 0xa000cb3c 0x00584355
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One
```

mossmann, 2022-11-21:
I recommend using a 20 dB attenuator between the two HackRFs to test this (and I believe the test script expects this).

mossmann, 2023-01-04:
I've observed (well under 1% of the time) the signal missing from sweep data in the second sweep as well as the first sweep, so I've started skipping the first three sweeps. (Skipping the third may be overkill, but I wanted to be safe.)

Perhaps this was the cause of these failures, although I was surprised to see it happen twice in a row: http://jenkins.greatscottgadgets.com/job/greatscottgadgets-organization/job/hackrf/view/change-requests/job/PR-1251/

martinling, 2023-11-07:
I wonder if this is a symptom of #1363.