greatscottgadgets/hackrf #1232: [Question]:  How do I get HackRF to run on Windows 10 in Pothos Flow?

rodgerwc, 2022-11-19:
### What would you like to know?

I am having difficulty getting HackRF to run on Windows 10 in Pothos Flow.

So far I have been able to install the USB driver, and verify that I can at least get a response from HackRF (or, at least the driver) using "SoapySDRutil.exe --find"

Found device 2
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 453c64c82369538f
  part_id = a000cb3c005e4353
  serial = 0000000000000000453c64c82369538f
  version = 2014.08.1

Then I tried:  SoapySDRutil.exe --probe='driver=hackrf'

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 'driver=hackrf'
←[1m←[33m[WARNING] SoapyVOLKConverters: no VOLK config file found. Run volk_profile for best performance.←[0m
←[1m←[31m[ERROR] sdrplay_api_Open() Error: sdrplay_api_Fail←[0m
←[1m←[31m[ERROR] Please check the sdrplay_api service to make sure it is up. If it is up, please restart it.←[0m
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.2; Boost_107500; UHD_4.0.0.HEAD-0-g90ce6062
←[1m←[31m[ERROR] SoapySDR::Device::enumerate(sdrplay) sdrplay_api_Open() failed←[0m

In this case there is mention of a VOLK config file, and an sdrplay_api service.   Where do I find the VOLK config file?  How do I see if the sdrplay_api service is running, and if it is not, how do I launch it?

I have also tried setting up HackRF sources and sinks in Pothos Flow. Where do I find the information needed to set them up? 

I am using Pothos Flow from  PothosSDR-2021.07.25-vc16-x64.exe

My intention is to get HackRF running using available binarys.  

Kind Regards,
Rodger

P.S. I did first look in https://github.com/greatscottgadgets/hackrf/issues to see if a similar issue was posted.   I could not find a similar issue.  I could not get into Discord since I do not own a mobile phone that is required for authentication.

straithe, 2022-11-22:
Can you provide a link to Pothos Flow? I don't think I'm familiar with it.

rodgerwc, 2022-11-23:
Straithe,

Thank you for responding to my lengthy question.

Pothos was referenced in  https://hackrf.readthedocs.io/_/downloads/en/latest/pdf/  on page 28.

When downloaded _here_  is pressed it opens:  https://downloads.myriadrf.org/builds/PothosSDR/

I downloaded:  PothosSDR-2021.07.25-vc16-x64.exe

I ran this executable and it unzipped into a PothosSDR subdirectory.

Now in PothosSDR, in the bin subdirectory, PothosFlow.exe can be found and then launched.

It appears to be based on GNUradio style system blocks.

I was able to get the HackRF transmitter to work nicely, as observed on a spectrum analyzer, using hackrf_transfer.exe (also in the above bin subdirectory).  I wrote a simple script in Python to then step over a range of frequencies so I could observe the transmitter's behavior as a function of frequency.  (See attached file)

This approach as a lot of latency.  Therefore I would like to try Pothos Flow.

Kind Regards,
Rodger

Python script using hackrf_transfer.exe:  Just change file extension to .py  ;-)

[hackrf_1.txt](https://github.com/greatscottgadgets/hackrf/files/10072188/hackrf_1.txt)

straithe, 2022-11-28:
Since it looks like your concerns are mostly with Pothos I suggest you reach out to the maintainers of that software for support. Their support page is here: https://github.com/pothosware/PothosCore/wiki/Support

rodgerwc, 2022-11-28:
Straithe,

Thank you for the suggestion, and the link.

Kind Regards,
Rodger

closed 2022-11-28 by rodgerwc