greatscottgadgets/hackrf #1234: [Tech Support]:

markl17, 2022-11-27:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

I tried the gnu radio 3.9 with soapy support which sucks to put it mildly so I downloaded 3.7 which has support for osmocom but it doesn’t seem to work at all even though I used the device driver and when I do hackrf_info it does work here is a link to what I am trying to accomplish 

### What outcome actually happened?

Nothing 
[image]

### What operating systems are you seeing the problem on?

Windows 11

### What is the output of ```hackrf_info```?

[image]

### Are you using any third-party software?

Gnu radio 

### Are you using any third-party hardware?

Hackrf and gnu radio to make signal generator

straithe, 2022-11-28:
It looks like your issues are with GNU Radio and not HackRF One. I suggest reaching out to the GNU Radio community: https://www.gnuradio.org/ as they have some excellent resources.

closed 2023-01-05 by straithe

markl17, 2023-01-05:
I did contact them they told me that it is hackrf issue https://ham.stackexchange.com/questions/21537/using-3-10-gnuradio-and-hackrf-to-make-a-signal-generator

straithe, 2023-01-05:
The link you have provided provides an answer using GNU Radio. Please implement the solution they have given you.

markl17, 2023-01-05:
he output that you see on that webpage is what I am getting here. Here I am posting the whole thing

markl17, 2023-01-05:
by the way I tried it with the original one with fried amp chip that still works and two new boards and none of them were recognized by gnuradio 3.10 which I got using conda install not brew. the hackrf stuff I installed using brew 
[image]
the one in the metal shield used to work on this mac before I it crashed and I had to reinstall the whole thing and I used conda not port because it is much quicker and simpler than recompiling the whole thing bringing in atleast 10 libraries which I seen port do any ideas

markl17, 2023-01-05:
this is what i wrote in the other forum 
there is a chain of events to unravel why hackrf is being ignored and zadig doesnt load it with the proper driver and then gnuradio produce the proper diagram. the weakest link in all of this win 11 ignoring the hackrf for some unknown reason – 
same thing seems to happen in mac os 10.15 catalina I installed gnuradio with "conda install gnuradio" and hackrf with "brew install hackrf". hackrf doesn't detect the board I have three of them, the original with burned amp chip which still functions and new once. the original used to work like a charm and still seems to work and boots up, and yet still not recognized by hackrf
[markl17](https://ham.stackexchange.com/users/23099/markl17)
 [Dec 16, 2022 at 14:34](https://ham.stackexchange.com/questions/21537/using-3-10-gnuradio-and-hackrf-to-make-a-signal-generator#comment37909_21538)

markl17, 2023-01-08:
My issue was resolved and it was because I didn’t research others reply’s. It was suggested that first and foremost one should check the cable. Most are made cheap, they don’t pass along anything but 5 volt pins