greatscottgadgets/hackrf #1241: How to use Opera Cake time mode within GNU Radio?

nikstp, 2022-12-15:
### What would you like to know?

Hi,

I'd like to develop a pseudo-Doppler application and I'm really not sure how I can access multiple ports within some programming environment with Opera Cake and HackRF.  I know you can configure the time mode thru the command line application, but how can I do that in the GNU Radio?

straithe, 2023-01-05:
Thank you for your question. There isn't a way to configure time mode from within GNU Radio, but you can configure time mode on the command-line and then launch the GNU Radio application. The time mode configuration will remain in effect until you unplug the HackRF or press the reset button or reconfigure with hackrf_operacake. The documentation at [hackrf.readthedocs.io/](https://hackrf.readthedocs.io/en/latest/opera_cake.html) highlights this by saying: "Once configured, an Opera Cake will remain in time mode until the mode is reconfigured or until the HackRF One is reset. You can pre-configure the Opera Cake in time mode, and the automatic switching will continue to work while using other software."

nikstp, 2023-01-05:
@straithe I've actually realized this two weeks ago, I just didn't get to trying it out it yet. I think I can make two sources in GNU Radio, each based on a different port, once the Opera Cake is in time mode. 

Once, if I do, get this working, I'll be happy to share an example!

straithe, 2023-01-12:
Thank you! Since it seems your question has been answered I will close this issue.

closed 2023-01-12 by straithe