greatscottgadgets/hackrf #1243: Extremely Poor RX Sensitivity

faultywarrior, 2022-12-17:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

I expect that the HackRF should be able to at least match cheapo no-name RTL dongles for strong-signal RX sensitivity and clarity.  For example, I would expect it to not have any problem receiving the NOAA Weather Radio station in my area.  Several RTL sticks and USRPs have no problem receiving it with their gains turned to less than 20db.

### What outcome actually happened?

The HackRF seems to have poor sensitivity and the resulting demodulation is very poor quality.  My settings are: LNA 24.0, AMP 8.0, VGA 33.0.  The received signal is very distorted, no matter what gain settings I try.  There is also a massive DC spike at my tuned frequency which I can't manage to compensate for.

### What operating systems are you seeing the problem on?

Ubuntu 22.04 AMD64

### What is the output of ```hackrf_info```?

hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c8319362cf
Board ID Number: 2 (HackRF One)
Firmware Version: 2022.09.1 (API:1.06)
Part ID Number: 0xa000cb3c 0x0055436a
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One

### Are you using any third-party software?

GQRX

### Are you using any third-party hardware?

Using the antenna cable/mag mount and extendable antenna from a NooElec NESDR Smart kit.

faultywarrior, 2022-12-17:
Closing this...issue appears to be related to GQRX trying to use the SoapySDR driver, which screws up the gain controls.

closed 2022-12-17 by faultywarrior

dmaltsiniotis, 2022-12-17:
Glad you figured it out though! The DC spike though, that will persist though. The way to get around that is to use offset tuning. Tune to some know offset of your frequency, then in software shift the signal back to center to avoid the DC spike.