greatscottgadgets/hackrf #1247: Wrong gain value(s) in FAQ section

Sasszem, 2022-12-23:
### What would you like us to add to or change about the HackRF One documentation?

In some sections of the FAQ, namely:
- [What gain controls are provided by HackRF?](https://hackrf.readthedocs.io/en/latest/faq.html#what-gain-controls-are-provided-by-hackrf)
- [Why is the RF gain setting restricted to two values?](https://hackrf.readthedocs.io/en/latest/faq.html#why-is-the-rf-gain-setting-restricted-to-two-values)

the value of the RF gain is incorrectly said to be 14dB. 

I also did a grep in the docs folder in the repo, and I only found this 3 mentions of 14 dB.

martinling, 2022-12-23:
Just confirming this and linking to where I wrote about it before: https://github.com/greatscottgadgets/hackrf/issues/1059#issuecomment-1060038293

Sasszem, 2023-01-05:
Closed via #1252

closed 2023-01-05 by Sasszem