greatscottgadgets/hackrf #1249: ofdm transceiver using hackrf one not successful.

reshma07mr, 2023-01-02:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

Trying to transmit and recieve files using ofdm transmission reception scheme via hackrf one.

### What outcome actually happened?

I tried to transmit and recieve files using ofdm transmission reception scheme via hackrf one. When the transmitter and receiver are connected using a virtual channel, the files are received as expected. But when the tranmsitter and receiver are connected via hackrf one, in the receiver side the data is not recovered correctly. I connected the hackrf ones using two coaxial cable, one cable for connecting the clocks and other for connecting the antennas.

### What operating systems are you seeing the problem on?

Tried on Windows 10 and Linux also.

### What is the output of ```hackrf_info```?

hackrf_info version : git-7047252
libhackrf version : git-7047252 (0.6)
Found HackRF
Index : 0
Serial number : 0000000000000000874461dc272d7557
Board ID Number : 2 (HackRF One)
Firmware Version : 2021.03.1 (API:1.04)
Part ID Number : 0xa000cb3c 0x007d4f5c

### Are you using any third-party software?

GNU Radio.

### Are you using any third-party hardware?

No

reshma07mr, 2023-01-12:
Do anyone experience dsimilar kind of things ? Any help provided on this will be greatly appreciated.

straithe, 2023-01-12:
Since your HackRF One detects the signal it seems your issue is in GNU Radio. I suggest you seek help in the GNU Radio community for more help: https://www.gnuradio.org/

closed 2023-01-12 by straithe

reshma07mr, 2023-01-13:
Thank you for your suggestion. I have seeked help in the GNU Radio community forum also. But #476 also refers the same, so do you think it is the issue with the GNU radio ?

straithe, 2023-01-13:
Yes.

reshma07mr, 2023-01-13:
Ok. Thanks