greatscottgadgets/hackrf #1256: UHF L-Band spectrum sweep

karriforesthermit, 2023-01-09:
### What would you like to know?

I am a beginner at SDR. I have new HackRF One running 2022.09.01 firmware. The following spectrum from the UHF L-Band is essentially from

hackrf_sweep -a 0 -l 16 -g 16 -f 1220:1250 -w 40000 -N5000

[image]

There are a number of signal peaks here. For example, the small one about 1227MHz,  in the middle of the first trough, which I assume is a GPS signal.

I have two questions:

1)  Why does the spectrum have those two unlikely looking troughs?

2) Why do I not see these troughs nor the big peak at 1225MHz when I use Gnuradio Companion to examine the spectrum around these frequencies? That is, using an Osmocom Source (20M sample rate, centre frequency at say 1226Mhz, bandwidth 40kHz) piped into a Frequency Sink.)

Is this something to do with the hardware configuration?

mmm545, 2023-01-09:
I believe those are LTE signals, GPS signals are much weaker than that and take more bandwidth

karriforesthermit, 2023-01-10:
OK. That makes sense.

martinling, 2023-01-10:
Something odd is definitely happening there but it makes a bit more sense if you know how the sweep mode works.

At each tuning step, we actually measure two 5MHz wide bands of spectrum, with a 5MHz gap in between them. So those two troughs are actually from the same batch of samples, covering 1225-1230 and 1235-1240 MHz. The big peak at  ~1225 MHz looks like it is actually just within the first band so probably part of the same problem.

When capturing these two bands, the HackRF will have been tuned to a centre frequency of 1232.5 MHz. If you tune to that centre frequency in GRC, do you also see anomalous results?

mossmann, 2023-01-10:
I think you have found a tuning bug. I can reproduce the 1225 MHz spike when I sweep over the same frequency range. Do you get different results if you start the sweep at 1199 MHz instead of 1200 MHz?

karriforesthermit, 2023-01-11:
Hello Mike, impressive suggestion. Starting the sweep at 1199MHz the big spike is greatly diminished, although the two troughs are still evident.

[image]

To Martin's suggestion, tuning to a centre frequency of 1232.5MHz does not appear to make any difference.  On the other hand, in GRC I find that even the "High" averaging in the Frequency Sink is not a sufficiently long period to identify any of the peaks seen in the sweeps.

karriforesthermit, 2023-01-13:
OK Thank you guys for your attention to this. I'll await the firmware update.
PS Mike, your YouTube lessons on HackRF are great and got me inspired to start on an SDR project. :)

straithe, 2023-04-02:
@karriforesthermit have you tried the latest update?