greatscottgadgets/hackrf #1257: Disable anti alias filter

WinstonVM, 2023-01-11:
### What would you like to know?

I've recently began researching on ways to listen on a larger bandwidth than the 20MHz bandwidth limit. And I stubmled accross this one recording of Mr. Ossmann in toorcon a few years ago whereby he disabled the anti aliasing filter and installing an external bandpass filter for intentional aliasing. However it was performed on the HackRF Jawbreaker. I'm working on a HackRF One, is it possible to disable the anti alias filter on the HackRF One?

martinling, 2023-01-11:
The baseband LPF is built into the MAX2837, all the settings for it are in the [datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/MAX2837.pdf). I'm not sure if there's any way to bypass it completely.

martinling, 2023-01-11:
Could you also clarify what you're hoping to achieve?

If you're wanting to capture e.g. two signals at once which are more than 20MHz apart, there may be tricks you can do in which you use the frontend mixer, with an LO frequency chosen to cause both signals to appear within the same 20MHz window in the IF band.

You may want to review the block diagram:
[image]

mossmann, 2023-01-12:
I've tried and failed to completely disable the anti-aliasing filter, but I have done intentional aliasing by setting the filter to the maximum bandwidth which is 28 MHz (or about 31 MHz with a firmware hack - see the MAX2837 data sheet).

WinstonVM, 2023-01-15:
> Could you also clarify what you're hoping to achieve?
>
> If you're wanting to capture e.g. two signals at once which are more than 20MHz apart, there may be tricks you can do in which you use the [...]

I'm trying to increase the maximum TX bandwidth to as large as I can.

WinstonVM, 2023-01-15:
> I've tried and failed to completely disable the anti-aliasing filter, but I have done intentional aliasing by setting the filter to the maximum bandwidth which is 28 MHz (or about 31 MHz with a firmware hack - see the MAX2837 data sheet).

I'm assuming it changes the register values using the hackrf_debug command? I've tried doing that, however, the data sheet doesn't seem to provide ample explanation on how to change the exact data bits itself.

martinling, 2023-01-23:
The relevant register settings are these ones, from page 20 of the [datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/MAX2837.pdf).

[image]

The way you get 31MHz is to select 28MHz and then enable the +10% tuning option.

It's not sufficient to do this with `hackrf_debug` though, because the firmware will overwrite what you set next time it writes to that register. You'd need to patch the firmware.

martinling, 2023-02-04:
You would need to make your changes to the firmware source code, [build the firmware](https://github.com/greatscottgadgets/hackrf/tree/master/firmware#readme) and then [update the firmware on your device](https://hackrf.readthedocs.io/en/latest/updating_firmware.html) to your modified version.

WinstonVM, 2023-02-05:
Noted, will attempt to do it. Will share an update here shortly.

WinstonVM, 2023-02-07:
Yes, it seems that the firmware hack method mentioned has worked. Although allow me to list down my steps if anyone reading this would like to replicate it.

1, I first cloned the hackrf firmware folder from this link <https://github.com/greatscottgadgets/hackrf/tree/master/firmware> using a Ubuntu virtual machine.
2. I traversed to the hackrf/firmware/common directory and edited the max2837.c file. 
3. On the function on Line 94, I appended the lines
|    set_MAX2837_FT(drv, MAX2837_FT_28M);
|    set_MAX2837_dF(drv, MAX2837_dF_10);
before the 
|    max2837_regs_commit(drv);
line.
4. After saving the file, I followed the guide on the build in this link <https://github.com/greatscottgadgets/hackrf/tree/master/firmware#readme>
5. And as for using the SPI Flash tool on the HackRF toolkit, I used this link <https://hackrf.readthedocs.io/en/latest/updating_firmware.html>

I would like to acknowledge that @martinling and @mossmann has helped me extensively throughout this goal of mine. They have provided the necessary guidance and resources for me to achieve it.

straithe, 2023-02-09:
Thank you for the update! Since the issue seems to be resolved I will be closing it now.

closed 2023-02-09 by straithe