greatscottgadgets/hackrf #1258: OFDM receiver using HackRF creates constellation plot even without the signal.

reshma07mr, 2023-01-12:
### What would you like to know?

Hi, I was working with OFDM data transmission using the hackRF one. Without when nothhing is transmitted, the receiver shows up a time domain signal and a QPSK constellation(4 point) plot at the end. Why is it so ?

Is it amplifying some signal from the laptop or is it something else ?

Do any one expeerienced like this, if so how can be eliminated ?
Thanks

reshma07mr, 2023-01-12:
Ok, but the thing is that , still #1249 IS UNRESOVED, even with the same configuration as provideed in the github account, I am not able to demodulate the signal.
Sorry, for the inconvenience caused in #1258, hereby attaching the screenshot of the flowgraph and the results I have seen in this [case.[](url)](
[image]
[image])

straithe, 2023-01-18:
I am closing this issue as it is not a HackRF focused issue.