greatscottgadgets/hackrf #1260: hackrf_sweep sometimes doesn't work on Raspberry Pi Zero 2 W

CodingSylvi, 2023-01-13:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

After about five to ten times of running the hackrf_sweep function on the Raspberry Pi Zero 2 W, it stops working and outputs:
0 total sweeps completed, 0.00 sweeps/second

My question is, if it's supposed to work on the Raspberry Pi Zero 2 W.
(I have tried the same thing on the Raspberry Pi 4 B, and it doesn't work.)

### What are the steps to reproduce this?

run   $ hackrf_sweep -f 10:6010 -B -w 1000000 -l 16 -g 24 -1    repeatedly

### Can you provide any logs? (output, errors, etc.)

$ uname -a
Linux raspberry pi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux

$ hackrf_info
hackrf_info version: git-b6acd66c
libhackrf version: git-b6acd66c (0.7)
Found HackRF
Index: 0
Serial number: 00000000000000000f77c60dc279182c3
Board ID Number: 2 (HackRF One)
Firmware Version: git-6f26b74e (API:1.07)
Part ID Number: 0xa000cb3c 0x00544754
Hardware Revision: r7
Hardware appears to have been manufactured by Grat Scott Gadgets.
Hardware supported by installed firmware:
	HackRF One

CodingSylvi, 2023-01-18:
I checked, and I already have the newest `libusb` version installed.
Has somebody already tested `hackrf_sweep` on the relatively new Raspberry Pi Zero 2 W.
Is it possible that the problem is related to the usb connection (speed, supply voltage, ...)
Maybe the problem is my fault:
I am calling `hackrf_sweep` as a subprocess in a python program repeatedly.
$ hackrf_sweep -f 10:6010 -B -w 1000000 -l 16 -g 24 -1

martinling, 2023-01-18:
Running it repeatedly shouldn't be an issue. This is exactly what [QSpectrumAnalyzer](https://github.com/xmikos/qspectrumanalyzer) does behind the scenes, for instance. Could you try running that and see if it works?

USB problems could certainly be involved. A HackRF One is a fairly demanding USB 2.0 device: it draws close to the maximum allowed current, requires high speed operation, can use nearly all the available bandwidth, and wants as little latency as possible. So it can certainly show up problems on platforms with poor USB implementations. Poor quality USB cables are also sometimes a cause - can you try a different one?

A useful test of the USB stack is to just try RX mode with buffer stats enabled:
`hackrf_transfer -r /dev/null -s 10e6 -B`

This should print lines like:
`19.9 MiB / 1.000 sec = 19.9 MiB/second, average power -35.4 dBfs, 5280 bytes free in buffer, 0 overruns, longest 0 bytes`

Then try increasing from the 10Msps sample rate `10e6` and see where you start getting overruns. Many hosts will get some if you go up to 20Msps but lower rates should be OK. However, even if the host is struggling to maintain throughput, this should not cause sweeps to stop completely, they will just take longer. Something else is going wrong there.

Another useful bit of info would be to retrieve the M0 state with `hackrf_debug -S` after a sweep run fails:

This will print something like:
```
M0 state:
Requested mode: 0 (IDLE) [complete]
Active mode: 0 (IDLE)
M0 count: 30046496 bytes
M4 count: 30031872 bytes
Number of shortfalls: 0
Longest shortfall: 0 bytes
Shortfall limit: 0 bytes
Mode change threshold: 30048256 bytes
Next mode: 0 (IDLE)
Error: 0 (NONE)
```

If however the counts are all zeros, that would indicate that the HackRF had been reset while running the sweep, which could be due to not getting enough power from the USB port.

dmaltsiniotis, 2023-01-19:
Running many sweep one-shots (starting and stopping the device in rapid succession) has been known (#916) in previous versions of the firmware/hackrf libs to cause the HackRF One to get into a state where it needs to be reset. This looks awfully familiar to that issue. Also, unless I am mistaken, I think QSpectrumAnalyzer works on the stream of continuous data coming from hackrf_sweep, and not in a loop calling start() and stop() in rapid one-shot successions.

However that issue was resolved with the refactor of how USB transfers are handled with libusb, so unfortunately I'm at a bit of a loss as for additional recommendations beyond those already suggested.

straithe, 2023-04-07:
I'm closing this as there hasn't been a response in over 30 days. Please re-open this issue or open a new one if you still need assistance.

khs4github, 2023-06-04:
My colleague CodingSylvi raised the technical support request "hackrf_sweep sometimes doesn't work on Raspberry Pi Zero 2 W #1260". I should finish the content.

As recommended on January 18th I tested the USB stack by "hackrf_transfer -r /dev/null -s 10e6 -B".
10e06, 11e06, 12e06 and 13e06 work fine without overruns. From 14e06 and above it doesn't work anymore.

Is there a simple way to limit the sampling rate?
Thank you for your support!

martinling, 2023-06-05:
In sweep mode the sampling rate is fixed at 20Msps.

However, during sweep the device is only sampling 1/3 of the time, so the USB data rate in sweep mode is only 6.66Msps. Given that you've tested up to 13Msps successfully with `hackrf_transfer`, I don't think USB throughput capability of the Pi is your problem.

Your problem might be related to the way that the `hackrf_sweep` subprocess is created and used by your Python program. If the output stream is not being read promptly enough, or is directed to a pipe with an insufficient buffer size, then the `hackrf_sweep` process could be blocked in a print call, preventing it from reading further data from the device.

Try simply running your `hackrf_sweep` command in a loop from the shell:

```$ while 1; do hackrf_sweep -f 10:6010 -B -w 1000000 -l 16 -g 24 -1; done```

Do you still see failures then? If not, then the problem is with the Python program.

khs4github, 2023-06-13:
The situation is:
* Raspberry Pi Zero 2W
* Raspberry Pi OS with desktop, May 3rd 2023, 64-bit (6.1)
* Connections:
** microUSB/PWR IN for Power
** microUSB/USB 2.0 OTG: HackRF One with an Adapter to USB A
** MINI HDMI PORT: not used

* WLAN with a fixed IP-Adress
* Putty as Terminal

What happens:
* "hackrf_transfer -r /dev/null -s 10e6 -B" => 0 overruns
* "hackrf_transfer -r /dev/null -s 12e6 -B" => 0 overruns
* "hackrf_transfer -r /dev/null -s 14e6 -B" => overruns!

* "hackrf_info" see file below.
* "hackrf_sweep -f 10:6010 -w 1000000 -l 16 -g 24 -1" for 2...3 times
  => "0 total sweeps completed, 0.00 sweeps/second" see file below.
* "hackrf_debug -S" see file below.

* Using htop: There seems to be enough memory and CPU.

File:
[TerminalRPI_01.txt](https://github.com/greatscottgadgets/hackrf/files/11739402/TerminalRPI_01.txt)

What could my next steps be?
+ Raspberry Pi OS Lite?
+ 64-bit => 32-bit
+ Installing from github "cmake .." brings the following warning:
  CMake Warning (dev) at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (LIBUSB)
  does not match the name of the calling package (USB1).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
  ???
+ I could document the entire installation in detail?

What do you think should happen now?
Best regards!

muaddib1984, 2023-08-07:
@martinling  I can confirm this behavior on both RPI zero 2W and RPI 4 (8GB), the process hangs for a while if the bin width is low.
I'm using a fresh Ubuntu22.04 server image on both platforms. I attempted running hackrf_sweep in one-shot mode with 1GHz sweep range (100:1100) and bin widths of 10k,50k,100k,1M, for the 10/50k widths, after approximately 20-30 seconds of waiting, it finally runs. Time to start is faster with wider bin widths. For the slower cases, during setup one core is at 100% while until it runs. Once it gets going, multiple cores are engaged. Running with number of sweeps set to 10, htop shows the same single core maxed out before the sweep starts, followed by moderate (and multi-core) CPU usage. 
I ran strace as well (don't have the exact output in front of me) which shows the process hanging right after ```prlimit...``` is displayed in the output. 
HackRF software is the latest master branch (as of about a week ago).

Let me know if I can help test any fixes (if there are any).

khs4github, 2023-08-22:
If I can help in any way, e.g. test with a modified version, please let me know!
In any case, thank you very much!

martinling, 2023-10-27:
Thanks to @ai6aj there is now a proposed fix for this on PR #1366. I don't have an RPi Zero 2 W to test with, so if anyone here is able to test the git branch from #1366 and see if it fixes the issue on that platform, that would be much appreciated.

ai6aj, 2023-10-27:
I'm quite interested to see if this fixes the issue as well.  I don't have a Pi Zero 2W but I developed/tested on a RPi 4 and I'm amazed at how incredibly slow planning is on that platform.  The Zero 2W has a slower processor and is bound to be even worse.  

If you use the -W option to store FFTW wisdom, subsequent runs are pretty fast.

khs4github, 2023-10-27:
I will try to run the tests on Raspberry Pi 4B and Raspberry Pi Zero 2W with buster, bullseye and bookworm in the next 2 weeks.

ai6aj, 2023-10-27:
FFTW's documentation suggests using the "estimate" plan if you need a plan to execute quickly.  I just tried **-P estimate** _without_ using wisdom and it starts up quite quickly on the RPi 4.  If the FFT it produces works fast enough on the Pi 2W I suggest making it the default.

khs4github, 2023-10-31:
So far I have assembled the software as follows:
- sudo apt install cmake build-essential pkg-config git
- sudo apt install libusb-1.0-0-dev usbutils libfftw3-3 libfftw3-dev ... J
- sudo apt update
- sudo apt upgrade
- mkdir LPSpec
- cd LPSpec
- git clone http://github.com/mossmann/hackrf.git
- cd hackrf
- git checkout v2023.01.1
- cd host
- mkdir build
- cd build
- cmake ..
- make
- sudo make install
- sudo ldconfig
- python3 --version
- hackrf_info
- hackrf_sweep -f 10:6000 -w 100000 -1
- hackrf_sweep -f 10:6000 -w 100000

What needs to be done to incorporate the changed release?

martinling, 2023-10-31:
I think this should be sufficient:
```
git remote add ai6aj https://github.com/ai6aj/hackrf.git
git fetch ai6aj
git checkout ai6aj_20231023_1366
make
sudo make install
```
Then re-run `hackrf_sweep` to test.

khs4github, 2023-11-02:
**Raspberry PI 4B/4GB/bullseye/64-bit:**
Creating the new version of hackrf_sweep works as described:
- sudo apt install cmake build-essential pkg-config git
- sudo apt install libusb-1.0-0-dev usbutils libfftw3-3 libfftw3-dev ... J
- sudo apt update
- sudo apt upgrade
- mkdir LPSpec
- cd LPSpec
- git clone http://github.com/mossmann/hackrf.git
- cd hackrf
- git checkout v2023.01.1
- git remote add ai6aj https://github.com/ai6aj/hackrf.git
- git fetch ai6aj
- git checkout ai6aj_20231023_1366
- cd host
- mkdir build
- cd build
- cmake ..
- make
- sudo make install
- sudo ldconfig
WiFi was switched off to obtain comparable results, the result seems to be good:
"hackrf_info" is OK.
"hackrf_sweep -f 10:6000 -w 100000 -1" is OK
"hackrf_sweep -f 10:6000 -w 100000" is OK even after more than 10 hours, the memory consumption remains constant (approx. 424M), there is no swapping (0K/100.0M).

**Should certain numbers be determined for comparison ("hackrf_sweep -f ???") or is it enough for hackrf_sweep to run and the result to be OK?**

martinling, 2023-11-02:
That looks like it's fixed! The swept frequency range shouldn't affect this so that test looks sufficient to me.

Thank you @khs4github for your work on testing this, and @ai6aj for the fix.

I'm going to approve PR #1368 now, and once that's merged this issue will be automatically closed.

closed 2023-11-02 by mossmann

khs4github, 2023-11-07:
**Raspberry PI Zero 2 W/bullseye/64-bit:**
I have now created and tested the new release on the Raspberry Pi Zero 2 W exactly as described above for the Raspberry Pi 4B. In addition, it makes sense to use a swap file with 2048 MB on the Raspberry Pi Zero 2 W.
WiFi was switched off to obtain comparable results, the result seems to be good:
"hackrf_info" is OK.
"hackrf_sweep -f 10:6000 -w 100000 -1" is OK
"hackrf_sweep -f 10:6000 -w 100000" is OK even after more than 10 hours, the memory consumption remains constant.
On the Raspberry Pi 4B there are approx. 51000 sweeps in 37600 seconds, on the Raspberry Pi Zero 2 W there are approx. 17800 sweeps in 36000 seconds.

On the weak Raspberry Pi Zero 2 W, the 4 cores of the CPU are constantly occupied at approx. 100%, 55%, 25% and 5%.
This means that hackrf_sweep now runs (albeit correspondingly slowly) on the Raspberry Pi Zero 2 W.

When will there be a new release of the software including this fix?
Thank you very much!

perfectglitch, 2023-12-03:
A dirty hack that worked for me on Raspberry 3B with Python (I do this after every time after hackrf_sweep subprocess has finished):

`pip install pyusb`

`
from usb.core import find as finddev
dev = finddev(idVendor=0x1d50, idProduct=0x6089)
dev.reset()
`

commit 799d9ad1 (2023-11-02): Run fftw_execute() once on an empty buffer (Fix #1366), add option to save wisdom and specify plan (#863) (#1368)

* Call fftw_execute once prior to starting IQ streaming
* Add options to use/save to an FFTW wisdom file and specify plan
* Fix bug where using -P would also enable one-shot mode

Fixes #1366
Fixes #1260
Addresses #863