greatscottgadgets/hackrf #1261: Is there a low power mode for hackRF?

alexmealster19, 2023-01-13:
### What would you like to know?

I'm working with a hackRF to run GNSS-SDR for a GPS fix. I have it working but the hackRF is consuming a lot of power. I was wondering if there was any way to configure the hackRF to potentially use less power. I am using an external oscillator and do not need to transmit.

straithe, 2023-01-18:
No, HackRF One does not have a "low power" mode and there isn't really anything you can change that will make a big difference in how much power HackRF One uses. There are low power GPS receiver implementations, HackRF One is not one of them. Thanks for the question!

straithe, 2023-02-09:
I'm closing this issue as the question has been answered.

closed 2023-02-09 by straithe