greatscottgadgets/hackrf #1264: TVS Diode/ Board issue

pat28475, 2023-01-16:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

Hello,

I recently had an issue with my Hack RF and it seems the center conductor for the coax connector became loose and took the pad off the board with it. During my attempt to re flow the solder, I believe the TVS Diode fell off and I cannot find it. It still receives signal by jumping the trace further up the line, I did not check transmitting. 
[image]
[image]

### What outcome actually happened?

I would like some help either replacing the broken and missing diode or some guidance on what to do. 

Thank you!

### What operating systems are you seeing the problem on?

Windows 11, various SDR programs

### What is the output of ```hackrf_info```?

N/A

### Are you using any third-party software?

N/A

### Are you using any third-party hardware?

EMI Shield

martinling, 2023-01-16:
The TVS diode is connected between signal and ground, so its absence won't stop the HackRF from working, but it does leave the RF port exposed to ESD damage.

The original part is [Murata LXES15AAA1-100](https://media.digikey.com/pdf/Data%20Sheets/Murata%20PDFs/LXES15AAA1-100.pdf). That part is now obsolete but you can use any similarly specified one. The key requirement is that it should have very low capacitance (the original is 0.05pF), otherwise it will degrade performance at high frequencies.

pat28475, 2023-01-17:
Perfect! Thank you!

pat28475, 2023-01-18:
> The TVS diode is connected between signal and ground, so its absence won't stop the HackRF from working, but it does leave the RF port exposed to ESD damage.
>
>
>
> The original part is [Murata [...]

I found this one, just wondering what are your thoughts. Thank you again

https://m.littelfuse.com/~/media/electronics/datasheets/pulseguard_esd_suppressors/littelfuse_pulseguard_pgb1_datasheet.pdf.pdf

martinling, 2023-01-18:
The capacitance on that looks fine. It doesn't supress an ESD spike as much as the original part but should still provide some protection. But this is an area where there are no perfect solutions - there's a limit to how much can be done if you still want to be able to get 6GHz signals through.

pat28475, 2023-01-19:
Okay I will keep looking and make a decision soon. Thank you

straithe, 2023-02-23:
I'm closing this as there hasn't been an update in over 30 days. Please re-open this issue or open a new one if you still need assistance.

closed 2023-02-23 by straithe