greatscottgadgets/hackrf #1265: Reconnection/Reset hackrfOne

alexmfz, 2023-01-18:
### What would you like to know?

Hello guys,

I would like to ask you if It is possible to reset by SW hackrf One or reconnect It through cmd in VM. 
Sometimes, the hardware dies st sweeping so It could be interesting to do that.

martinling, 2023-01-18:
Yes, you can reset an attached HackRF One with `hackrf_spiflash -R`.

alexmfz, 2023-01-18:
Nice!
So just executing that, the device is reattach and Will I be able to sweep again?

martinling, 2023-01-18:
Yes, although the sweep shouldn't be failing in the first place, so it would be good to hear more about the original problem. It sounds like you may be seeing similar problems to #1260 ?

alexmfz, 2023-01-18:
My problem is that sometimes hackrfOne doesnt sweep so I have to exit from execution with CTRL+C.

martinling, 2023-01-18:
Are you running the latest versions of the firmware and host software? If so, please open a bug report because that shouldn't happen.

closed 2023-03-21 by straithe