greatscottgadgets/hackrf #1266: Not showing up in Device Admin (not even in DFU mode)

GAsinPrieto, 2023-01-18:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

Y hoped to see the HackRF pop up in the device manager.

### What outcome actually happened?

Not recognized at anything, no connection.

### What operating systems are you seeing the problem on?

Windows 10 64b

### What is the output of ```hackrf_info```?

N/A

### Are you using any third-party software?

N/A

### Are you using any third-party hardware?

Portapack (HackRF works with Portapack, no USB either)

GAsinPrieto, 2023-01-18:
Y add a photo of the board.

[image]

straithe, 2023-01-18:
It sounds like you may need to try some different USB cables: https://hackrf.readthedocs.io/en/latest/troubleshooting.html#why-isn-t-my-hackrf-one-detectable-after-i-plug-it-into-my-computer

GAsinPrieto, 2023-01-19:
As stated in the troubleshooting documentation, I tried in all the USB2.0 (and 3.0) ports in my computer, which give a solid minimum of 500 mA, and all charge the device, but none is able to even try to recognize nor show it up at even unknown devices section in the device administrator. I have checked with all the cables I have around (data cables, not only-charging ones), and the behaviour is consistent: not even the sound of connection to Windows. Thus, I think the problem is not the cable.

The solders on the micro USB port seem to be solid and ok, and none of the components on the board seem to be burnt or blown off visually.

Thanks in advance for your help

dmaltsiniotis, 2023-01-19:
What happens if you restart the device in [DFU mode](https://hackrf.readthedocs.io/en/latest/updating_firmware.html#only-if-necessary-dfu-boot), does it show in device manager then? 

>To start up HackRF One in DFU mode, hold down the DFU button while powering it on or while pressing and releasing the RESET button. Release the DFU button after the 3V3 LED illuminates. The 1V8 LED should remain off. At this point HackRF One is ready to receive firmware over USB.

When in DFU mode, it should show up like this:
[image]

When loaded with proper firmware in standard mode with the USB light illuminated it should show up like this:
[image]

You mentioned a PortaPack and that the HackRF works with the PortaPack. I don't have any expereince with that add-on, but I know that some addon-ons require you to flash different/custom firmware on the device. If the HackRF one shows up device manager when in DFU mode, it may need to be re-flashed with the standard HackRF One firmware if you want it to operate that way.

GAsinPrieto, 2023-01-19:
Nope, it seems to enter in DFU mode (LEDs blink differently) but it is not showing up in the device admin either. The behaviour is also consistent with and wothout the Portapack connected to the HackRF, i.e., leaving the HackRF as default.

straithe, 2023-01-19:
What version of PortaPack do you have?

GAsinPrieto, 2023-01-19:
Portapack H2 it is

GAsinPrieto, 2023-01-19:
From the complete ignorance of the working principle on U15, may this chip be interefering with the correct working of the USB peripheral in the microcontroller? IMHO it would be weird, as it seems OK visually (I have seen photos of that IC blown up, but well, that does not mean anything) and at least I guess that sometimes I may get to at least suddenly see the device trying to connect, but it is not the case.
Furthermore, I have seen somebody at Discord with a similar problem that did continue having the same issue (hardware not recognized at all by the computer) after removing U15, so I do not know if that is the correct direction.

martinling, 2023-01-19:
U15 is a TVS diode array ([datasheet](https://www.vishay.com/docs/84145/vbus54cvhsf.pdf)) which is there to protect the USB lines from electrostatic discharge events. The USB signals pass by it, not through it. It's extremely unlikely to be the source of a problem; it's there to protect everything else.

GAsinPrieto, 2023-01-19:
Thanks for the info, I was just guessing out of others experience of burnt U15 and USB coming back to life when removed or replaced, but it's true it was not this but the old component reference (old: LXES1TBCC2-004; new: VBUS54CV-HSF-G4-08).

straithe, 2023-01-21:
Again, I would suggest trying a few more USB cables.

GAsinPrieto, 2023-01-24:
On a trip now, will test it when I'm back with another couple cables and let you know. Thanks for the help

GAsinPrieto, 2023-01-27:
Hi, tried again with all the cables and some more, but still no signal in either modes (HackRF and DFU). So, I just wonder:
- Is it possible that the firmware is not correctly updated, and thus HackRF mode is not correctly triggering USB because of this?
- Where is DFU firmware or bootloader stored? Could it be also broken due to a bad update or something?

I do not know if these questions make any sense, and if so, maybe updating the full microcontroller memory via JTAG connector could solve the issue...

Thanks in advance

mmm545, 2023-01-27:
The DFU is built into the LPC4320 boot ROM. So no, a firmware upgrade/downgrade won't affect it
[image]

martinling, 2023-01-27:
The DFU bootloader is hardwired in mask ROM on the microcontroller. It's not possible to change it at all.

The fact that you get LEDs lighting up differently when starting with the DFU button pushed confirms that the MCU is getting power, distinguishing between startup modes and executing some code. That should more or less guarantee that when started in the DFU mode it should be correctly attempting to present itself on the USB bus.

Since you've already tried multiple host ports, and assuming those include cables with known good power & data lines, the area within which there must be a fault is mostly limited to the MCU, the USB connector and the traces in between. Without mechanical damage, which I don't see in the photo, there's not really anything to go wrong on that path.

As such, I'm afraid the most likely explanation is that the USB frontend within the MCU is fried, which is a pattern we've seen a few times before. This can result from overvoltage conditions, including DC voltage fed into the RF port or excess voltage applied the USB VBUS line. Sometimes there are other associated symptoms, such as high current draw, and the MCU and/or the U21 regulator getting hot.

GAsinPrieto, 2023-01-27:
Hi, I have tried in a Linux computer, and similar results: lsusb does not seem to see any changes on connected devices. So I must agree with you: it seems that peripheral on the microcontroller is dead. I guess my only bet is to have the MCU replaced, isn't it?

straithe, 2023-02-09:
You can contact your reseller to see if they are able to replace the device for you.

GAsinPrieto, 2023-02-14:
I feel that is not a possibility anymore, it is a second hand device with a lot of time now. I will try the MCU replacement way I think, will let you all know.
Thanks for the help, I suggest adding some kind of protection to the USB HW if this is not an isolated case.

GAsinPrieto, 2023-05-21:
I think I have found a possible cause for this issue to happen: missconnecting the Portapack by 1 pin left or right, I mean, connecting the IDC connectors displaced, so that the pinout is not coincident. This is a problem as Portapacks do have the battery connected when attaching to the HackRF board. Does anybody feel this is a plausible cause?