greatscottgadgets/hackrf #1267: Requesting techical support on information regarding multicarrier transmission using hackrf one.

reshma07mr, 2023-01-18:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

no

### What outcome were you hoping for?

I would like to transmit 2 two carriers each of 1 MHz wide with a carrier separation of 2MHz using hackRF and GNU radio.

### What outcome actually happened?

I like to have the OFDM transmission using the hackrRF one in the GNU radio software. But right now, we are facing some issues such as, we would like to transmit 2 two carriers each of 1 MHz wide with a carrier separation of 2MHz. I haven't seen an option in the osmocom module available in the GNU radio for this purpose. 

Can you provide technical support on this situation, how can I configure the system for this purpose? 

Any help provide on this will be greatly appreciated.

### What operating systems are you seeing the problem on?

Linux OS 20.04 

### What is the output of ```hackrf_info```?

hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000874461dc272d7557
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x007d4f5c

### Are you using any third-party software?

GNU Radio, GQRX

### Are you using any third-party hardware?

no

martinling, 2023-01-18:
This is the third issue that you have opened in this repository asking for help implementing OFDM in GNU Radio.

As we have already explained, you are asking in the wrong place. Issues on this repository are for reporting problems with the HackRF hardware, firmware and low-level host software.

HackRF is a software defined radio. It's up to you to use software of your choice to generate the signals you want to transmit. For those new to SDR and GNU Radio, we have an [11-part introductory video course](https://greatscottgadgets.com/sdr/) to help people get started.

OFDM is an advanced modulation scheme, and implementing it with GNU Radio requires understanding a lot of basics first. If you are being asked to do this as part of a course assignment, you should ask your tutors for help.

Transmitting multiple carriers does not require special support from the SDR hardware or sink blocks you are using. You need to combine all the signals you want to transmit in a single sample stream, with sample rate that is sufficient to provide the neccessary total bandwidth. HackRF One supports up to a 20MHz sample rate. At 2MHz separation you could transmit up to 10 carriers simultaneously.

straithe, 2023-01-18:
@reshma07mr, as @martinling has stated, this repository is not the place for these questions. The team at Great Scott Gadgets does not have the time available to research this for you. As you've mentioned via emails to our team, in which you repeat the requests you've made here on GitHub, you have been unable to find an answer. This is a great opportunity for you to do your own primary research and experiments into finding an answer. Good luck!

closed 2023-01-18 by straithe