greatscottgadgets/hackrf #1268: command line "hackrf_operacake ***" not working

usa55b1c, 2023-01-23:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

setup antennas in operacake

### What outcome actually happened?

after entering the command in the command line, returned error

### What operating systems are you seeing the problem on?

raspberry pi4, be easy on me.  i'm a big windows user.  trying to get away from them.  

### What is the output of ```hackrf_info```?

usa55b1c@raspberrypi:~/Desktop/hackrf-2022.09.1/host/hackrf-tools/src $ hackrf_operacake
bash: hackrf_operacake: command not found

### Are you using any third-party software?

none, running through command line

### Are you using any third-party hardware?

none  
using offical hackrf, and offical opera cake.  no clones

usa55b1c, 2023-01-23:
thank you guys!!!

straithe, 2023-01-23:
What did you enter into the command line?

dmaltsiniotis, 2023-01-23:
With the limited information here, I have an idea on what might be going on: It _appears_ as if you have the hackrf package installed on your system via a package manager (apt) but perhaps you also downloaded the source from the git repository in an attempt to use the latest hackrf_operacake utility. This would explain by hackrf_info works but hackrf_operacake does not.

If my many assumptions are correct, there's a few things you'll need to do in order to get up and running:
1) Remove/uninstall the old hackrf libraries as they are a bit outdated. You can do this by running `apt-get remove hackrf`.
2) Follow the instructions [here](https://github.com/greatscottgadgets/hackrf/tree/master/host#how-to-build-the-host-software-on-linux) on how to compile the source code and install the latest tools, which includes the opera_cake command line utility.

Again, there are a lot of assumptions here, but if you complete those two steps, your system should be updated with the latest and greatest hackrf tools, which will include the hackrf_operacake utility.

usa55b1c, 2023-01-23:
I entered of course "hackrf_info" to make sure that the system sees the hackrf.  No problems.

Then I entered "hackrf_operacake -m frequency -f A1:0:4000" to set up the Operacake, and it returned "bash: hackrf_operacake: command not found"

Then I entered "hackrf_operacake -l" to see maybe the system did not see the Operacake, and returned "bash: hackrf_operacake: command not found" again.  

I did do a lot of crazyness in installing the hackrf board.  I probly did mess something up, and I'm new to this.  I will try that now, and report back later today.  thank you guys

usa55b1c, 2023-01-24:
so dmaltsiniotis, i opened the command line in copied pasted "apt-get remove hackrf", and got 

"E: Could not open lock file /var/lib/dpkg/lock-frontend - open (13: Permission denied)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), are you root?" error.

after looking the error up, and trying to stop some processes, i gave up and reloaded my pi4.  

attempted to do 2, and yes i'm new to this but..................  none of that worked, couldn't get past the first line "mkdir host/build"  

usa55b1c@raspberrypi:~ $ mkdir host/build
mkdir: cannot create directory ‘host/build’: No such file or directory

so i went back to hackrf.readthedocs software section, and used the "Installing From Source, Linux / OS X / *BSD: Building HackRF Software From Source"  it worked.  i could use "hackrf_info" to see the hackrf board.  it worked, i'm still not able to run any hackrf_operacake commands.  below are some of the errors i get, on what command i use.  

usa55b1c@raspberrypi:~ $ hackrf_operacake -m manual
hackrf_operacake: symbol lookup error: hackrf_operacake: undefined symbol: hackrf_set_operacake_mode

usa55b1c@raspberrypi:~ $ hackrf_operacake -l
Opera Cakes found: 
hackrf_operacake: symbol lookup error: hackrf_operacake: undefined symbol: hackrf_get_operacake_mode

usa55b1c@raspberrypi:~ $ hackrf_operacake
Specify either list, mode, or GPIO test option.

Usage:
	-h, --help: this help
	-d, --device <n>: specify a particular device by serial number
	-o, --address <n>: specify a particular Opera Cake by address [default: 0]
	-m, --mode <mode>: specify switching mode [options: manual, frequency, time]
	-a <port>: set port connected to port A0
	-b <port>: set port connected to port B0
	-f <port:min:max>: automatically assign <port> for range <min:max> in MHz. This argument can be repeated to specify a list of ports.
	-t <port:dwell>: in time mode, dwell on <port> for <dwell> samples. Specify only <port> to use the default dwell time (with -w). This argument can be repeated to specify a list of ports.
	-w <n>: set default dwell time in samples for time mode
	-l, --list: list available Opera Cake boards
	-g, --gpio_test: test GPIO functionality of an Opera Cake

dmaltsiniotis, 2023-01-25:
Hi @usa55b1c Nice job compiling the hackrf tools from source, you've made some great progress so far. It looks you're now running into an compatibility issue with hackrf_operacake (from the error hackrf_operacake: undefined symbol: hackrf_get_operacake_mode) because the first apt command to uninstall the old tools did not work and the new tools are mixing and matching with the old ones. In Linux, when uninstalling or installing system wide things, you'll often need to perform these tasks as an administrator. In order to do this, you'll need to put the command `sudo` in front of the command that needs administrative privileges. So the command to uninstall the old tools becomes `sudo apt-get remove hackrf`. That will remove the old tools. Once you do that you may have to run `sudo ldconfig` again. Give that shot and see if resolve the hackrf_operacake command issue.

usa55b1c, 2023-01-25:
i got the remove hackrf to work (thank you very much for the class.  Im in the process of going through websites, and getting my youtube masters degree on Linux hahahhaa)

the below are the instructions from hacdrf.readthedocs page.  

Installing From Source[¶](https://hackrf.readthedocs.io/en/latest/installing_hackrf_software.html#installing-from-source)
Linux / OS X / *BSD: Building HackRF Software From Source[¶](https://hackrf.readthedocs.io/en/latest/installing_hackrf_software.html#linux-os-x-bsd-building-hackrf-software-from-source)
Acquire the source for the HackRF tools from either a [release archive](https://github.com/mossmann/hackrf/releases) or git: git clone https://github.com/mossmann/hackrf.git

Once you have the source downloaded, the host tools can be built as follows:

cd hackrf/host
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

did not work, i get the same errors.  

those are the same instruction that i followed after reloading the OS and was also getting the same error.  Maybe because i'm using a raspberry pi4 to do all this on???

dmaltsiniotis, 2023-01-26:
Can you post the error you're getting? You should be able to build on a pi4.

usa55b1c, 2023-01-26:
usa55b1c@raspberrypi:~ $ hackrf_operacake -m manual
hackrf_operacake: symbol lookup error: hackrf_operacake: undefined symbol: hackrf_set_operacake_mode

usa55b1c@raspberrypi:~ $ hackrf_operacake -l
Opera Cakes found:
hackrf_operacake: symbol lookup error: hackrf_operacake: undefined symbol: hackrf_get_operacake_mode

usa55b1c@raspberrypi:~ $ hackrf_operacake
Specify either list, mode, or GPIO test option.

Usage:
-h, --help: this help
-d, --device : specify a particular device by serial number
-o, --address : specify a particular Opera Cake by address [default: 0]
-m, --mode : specify switching mode [options: manual, frequency, time]
-a : set port connected to port A0
-b : set port connected to port B0
-f port:min:max: automatically assign for range min:max in MHz. This argument can be repeated to specify a list of ports.
-t port:dwell: in time mode, dwell on for samples. Specify only to use the default dwell time (with -w). This argument can be repeated to specify a list of ports.
-w : set default dwell time in samples for time mode
-l, --list: list available Opera Cake boards
-g, --gpio_test: test GPIO functionality of an Opera Cake

martinling, 2023-01-26:
If you run `which hackrf_operacake`, that will show you the path to the binary, which I would expect to be at `/usr/local/bin/hackrf_operacake`.

You can then use `ldd` with the resulting path to see which libraries are being used when you try to run that command, e.g:
```
$ ldd /usr/local/bin/hackrf_operacake
	linux-vdso.so.1 (0x00007ffde31bc000)
	libhackrf.so.0 => /usr/local/lib/libhackrf.so.0 (0x00007fcd95b7f000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fcd9599e000)
	libusb-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007fcd95980000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fcd95bcb000)
	libudev.so.1 => /usr/lib/x86_64-linux-gnu/libudev.so.1 (0x00007fcd95958000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fcd95953000)
```

Your problem appears to be that your system is defaulting to an out of date `libhackrf.so.0` somewhere on the system. The `ldd` output should reveal where that is so that you can remove it.

 The places that the system will look for libraries are configured in `/etc/ld.so.conf`. Make sure that you have `/usr/local/lib` in there somewhere as that is where `make install` will have put the ones you built.

usa55b1c, 2023-01-26:
yes, found the libhackrf.so.0       

remove it, and done?

do i need to replace it with an updated version.??

martinling, 2023-01-26:
When you built the host software and ran `sudo make install`, that should have put the newly built library at `/usr/local/lib/libhackrf.so.0`, as well as putting the command line tools like `hackrf_operacake` in `/usr/local/bin`. So everything you need should already be there. The point of the `ldd` was to figure out why the system seems to be using the wrong version of the library. Rather than just deleting it, look at what path it's using and why it's wrong. My guess is that you still have an old version of the `libhackrf0` package installed, which would be at `/usr/lib/libhackrf.so.0` and which you can remove with `sudo apt remove libhackrf0`. But that doesn't explain why the version you built yourself isn't automatically taking precedence, because normally `/usr/local/lib` has priority over `/usr/lib` in the `/etc/ld.so.conf` configuration. Did something not succeed when you built and installed everything?

usa55b1c, 2023-01-26:
martinling, everything went perfect during the install.  no errors, nothing.  

i ran "git clone https://github.com/mossmann/hackrf.git" 

it created the files 

i ran the below exact commands
cd hackrf/host
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

after running "which hackrf_operacake" the below is the output

linux-vdso.so.1 (0xbef7f000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0xb6f59000)
	libhackrf.so.0 => /lib/arm-linux-gnueabihf/libhackrf.so.0 (0xb6f2f000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6ddb000)
	libusb-1.0.so.0 => /lib/arm-linux-gnueabihf/libusb-1.0.so.0 (0xb6db3000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6d87000)
	/lib/ld-linux-armhf.so.3 (0xb6f6e000)
	libudev.so.1 => /lib/arm-linux-gnueabihf/libudev.so.1 (0xb6d53000)

martinling, 2023-01-26:
OK, so on your system the problematic library isn't in `/usr/lib` but besides that it's the same problem I described. You have an `/lib/arm-linux-gnueabihf/libhackrf.so.0` which is an old version, probably installed as part of a `libhackrf0` package on your distribution. You should remove that package to get rid of it, and you may also need to add `/usr/local/lib` to your `/etc/ld.so.conf`.

usa55b1c, 2023-01-29:
ok update jents, 

i went in and removed a "software defined radio peripheral"  package and it works fine.  no errors, nothing.  below is the output.  yes, martinling it was part of a libhackrf0 package on one of my distributions.  

ackrf_info version: git-6f1cee1f
libhackrf version: git-6f1cee1f (0.8)
Found HackRF
Index: 0
Serial number: 0000000000000000325866e6228a5823
Board ID Number: 2 (HackRF One)
Firmware Version: 2022.09.1 (API:1.06)
Part ID Number: 0xa000cb3c 0x00584f58
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One
Opera Cake found, address: 0

only problem now is that CubicSDR does not want to work, unless the same package is installed. go figure.  ill work on this and comb the web for a fix.  if i cant find it, guess i'll be back.

martinling, 2023-01-29:
Glad you got it working!

closed 2023-01-29 by martinling

usa55b1c, 2023-01-29:
for you alls information, just in case someone else has this problem.

after i removed the package with the old version, all the hackrf commands worked, but CubicSDR did not (CubicSDR would not see my Hackrf).

I removed CubucSDR and reloaded it.  Maybe CubicSDR was still pointing to the old version and did not want to work.  After reinstalling CubicSDR with the correct package with the updated libhackrf0, CubicSDR works.  

thank you guys, for helping the newbe!!!!!!!!