greatscottgadgets/hackrf #1269: Was there a resolution to #574  specifically sweep transmit

smileyrekiere, 2023-01-26:
### What would you like to know?

Was there a resolution to sweep for transmit #574 (also mentioned in #364, closed and tracked by #574)?  

I've used gnu radio companion to create a python script to modify the frequency during transmit yet does not have the performance needed for tx sweep.  So curious if there is a TX sweep as mentioned in #574.

There was a program hrfJam , posted in #574, which actually performs a tx sweep. Yet no source.  (tested it out and used the hackrf spectrum to see how is performed).

martinling, 2023-01-26:
There's no special features for this at present. Issue #364 was left open to track the idea.

The way to approximate a TX sweep in software is to generate a linear chirp across N MHz of baseband bandwidth, transmit that at your starting frequency, then retune upwards by N MHz and repeat. I have not looked at the program you mention but this is probably what it does. Note that doing this will not result in a continuous sweep, because some time is required for the program to stop transmitting, retune the HackRF, and start transmitting again at the next frequency.

When you say you need more performance, what sweep rate do you actually need, and how continuous do you need the sweep to be? What are you trying to do?

With the most recent 2022.09.1 release, the time needed to stop and restart short TX/RX operations has been significantly improved, so I would suggest having a close look at whether you can already do what you need through software.

Things could be made faster still by moving the retuning process into the firmware, as is done for the RX sweep mode. But even from firmware it would not be possible to achieve a continuous sweep, as there is still some time required for retuning: the MCU has to send the necessary commands to the mixer and transceiver chips, and then wait for their PLLs to settle. This takes a millisecond or so during which it's not possible to transmit.

I expect the main usefulness of a TX sweep mode would be to use it in combination with a second HackRF running a corresponding RX sweep, with the two locked to a shared clock source and using the hardware sync mechanism. The two could then be used together for various measurements. But you can already do this in software, if you can live with the time needed to manage stopping, tuning and restarting from the host.

smileyrekiere, 2023-01-26:
Question: "When you say you need more performance, what sweep rate do you actually need, and how continuous do you need the sweep to be? What are you trying to do?"

Try to implement Long Pulse Radar Test Waveform 
Pulse Width (micro seconds)  50-100 and a sweep width of 5 - 20 Mhz.  A continuous sweep would be optimal, yet need to see if can work with hardware available (hackrf) or show it will not work.

[image]

The python implementation did not meet the completion of the sweep in 50-100 micro seconds.
In python was using the run method as compared to start, stop , wait.  I did  see that when running in a continuous mode (or repeat) the frequency changes were accepted and changed.

martinling, 2023-01-26:
If you only need to sweep 20MHz, there's no continuity issue and no need to retune the HackRF - just set your sample rate to 20MHz, tune the HackRF to the centre frequency you want, and generate the samples for a sweep that goes from -10MHz to +10MHz.

smileyrekiere, 2023-01-26:
Thank-you,  so the sweep is modulated on the centre frequency.  

Re-looked at the sample plots which aligns with your recommendation of -10MHz to +10MHz. (graph from Vector Signal Analyzer) .  As you pointed out the centre frequency is not re-tuned.  

Currently I am creating a sample 'sweep vector' using the chirp (sweep) method from scipy using linear and quadratic. 
If there is a preferred method for creating a sweep from -10MHz to +10 MHz, I would be interested in knowing.  

Will also seeing if I can get the second hackrf to demodulate the sweep (chirp).

Here is the image referenced 
[image]

martinling, 2023-01-26:
Another way to produce the chirp would be a sawtooth signal source feeding into an FM modulator block in GNU Radio.

smileyrekiere, 2023-01-31:
Thank-you for the suggestion,  implemented the chrip with the sawtooth signal source feeding into an FM modulation block with GNU radio companion and verified the chirp/sweep with a second hackrf.   Currently trying to determine the configuration to trigger a single sweep  of the sawtooth signal to allow for a single chirp across the 5-20 Mhz frequency range.
[image]

smileyrekiere, 2023-02-04:
The final method used was to have an input vector to osmocom Sink.  The values of the vector were a chirp generated by information described here:
https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.chirp.html.

closed 2023-02-04 by smileyrekiere