greatscottgadgets/hackrf #1270: Symbol lookup error: ./hackrf_sweep: undefined symbol

alexmfz, 2023-01-26:
### What would you like to know?

Hi guys,

First of all, thank you so much with the question you are answering me. I am really having a nice help with the problems.

I would like to ask you about an error that has appeared at using hackrf API in a new environment. The case is the following:
I developed my hackrf_sweep code for my application and everything went well. However, When I moved the .zip from another PC, I have encountered the following error:

**"Symbol lookup error: ./hackrf_sweep: undefined symbol....."**

Do you know where is the problem? Could It that be because there are different versions of libraries?

Thank you so much :)

straithe, 2023-01-26:
Could you please close this issue and open a new technical support issue?

closed 2023-01-26 by straithe

alexmfz, 2023-01-26:
Sure