greatscottgadgets/hackrf #1273: HackRF One not working transmission of GPS signal

Imtiaz08, 2023-02-01:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

hi there I am studying GPS anti spoofing for which I have to generate a spoofing signal and receive it on a GPS receiver. 

When I transmit the file I am getting this error 

[image]

### What are the steps to reproduce this?

I am using the pothosSDR hackrf_transfer file to transmit the signal

Imtiaz08, 2023-02-09:
> Based on your image it looks like the error is telling you that you reached the end of your input file. Is there anything else I can assist you with? 

Thank you so much for your kind response. 

Actually the file has a duration of 300 seconds. 

Also what does the amplitude inf means?

martinling, 2023-02-09:
If the duration doesn't match what you expect, you are either not giving the correct sample rate of the input data to `hackrf_transfer`, or the file does not actually have the length you expect.

The amplitude `-inf` appears because you are using an old version of hackrf_transfer which does not calculate the input data amplitude when transmitting. If you update to the 2022.09.1 release or later, a correct dBfs value will be displayed. The difference does not affect the actual transmission.

Imtiaz08, 2023-02-09:
> If the duration doesn't match what you expect, you are either not giving the correct sample rate of the input data to `hackrf_transfer`, or the file does not actually have the length you expect. [...]

I am generating the IQ data with 2.6e6 sampling frequency should I reduce it to 2e6?

Also how can I update the hackRF transfer

martinling, 2023-02-09:
You have to tell `hackrf_transfer` what the sample rate of your IQ data is. You would do that with the `-s 2.6e6` option.

I'm not sure how to go about updating the host tools within an existing Pothos installation. Ideally they would release an updated version of the PothosSDR distribution.

dmaltsiniotis, 2023-02-18:
The Pothos folks unfortunately appear to have stopped updating their SDR packages sometime in July 2021, with the last release being `PothosSDR-2021.07.25-vc16-x64.exe` for Windows. I've resorted to compiling the updated host tools to get the latest version, which isn't as painful as it sounds on windows.

closed 2023-03-21 by straithe